martin-ger/esp_wifi_repeater #282: make the ESP an AP and disable DHCP

odaiayoob, 2019-03-13:
Hello,

I need to use the esp as wifi extender or access point, without DHCP, how can i do that ?
i need to disable DHCP and set static IP, GW, subnet and DNS, then all IPs come from main DHCP

martin-ger, 2019-03-13:
Currently you cannot disable the DHCP server, but bou are free to choose static IPs in the APs subnet.

As the esp_wifi_repeater is in fact a (NAT-)Router, a central DHCP server won't work. Each AP has to span its own subnet and DHCP broadcast requests are not even forwarded across router boundaries.

odaiayoob, 2019-03-13:
Can you tell me where to start?

martin-ger, 2019-03-14:
If you can compile the sources with the esp-open-sdk, you can simply add a `wifi_softap_dhcps_stop()` after line 3688 in user_main.c.

But as said before: you can use static IPs in AP subnets even without modification: the DHCP server only assigns addresses in the range from 2-128. Anything from 129-254 is free for static assigned IPs.

zoomx, 2019-03-14:
I believe that odaiayoob mean to use IP in the same subnet of main DHCP.

odaiayoob, 2019-03-14:
@zoomx Correct,

I need to make my ESP an access point only and I need to set fixed IP for it from main DHCP, and any device connected to this Access point get the IP from main DHCP,

defaliz, 2022-01-18:
I just read this post, after write a new issue...
i want to do the same thing.. use only my main dhcp and disable esp dhcp...

geofrancis, 2022-01-18:
what your wanting is a transparent bridge mode so the extender is invisible on the network not doing nat.

martin-ger, 2022-01-18:
Transparent bridge does not work, sorry.

defaliz, 2022-01-19:
thank for your reply Martin and your great work !
regards