martin-ger/esp_wifi_repeater #283: 编译错误

ruozhuxuan, 2019-03-14:
您好，编译时出现如下错误怎么解决
build/app_app.a(user_main.o): In function `timer_func':
/home/huguangyuan/esp/esp_wifi_repeater-master/user/user_main.c:2954: undefined reference to `wifi_station_set_enterprise_identity'
/home/huguangyuan/esp/esp_wifi_repeater-master/user/user_main.c:2958: undefined reference to `wifi_station_set_enterprise_identity'
collect2: error: ld returned 1 exit status
Makefile:132: recipe for target 'firmware/0x02000.bin' failed
make: *** [firmware/0x02000.bin] Error 1
[image]

martin-ger, 2019-03-14:
You probably need a newer version of the SDK. Or, if you don't need WPA2 Enterprise, just set `WPA2_PEAP ` to 0 in user_config.h.

ruozhuxuan, 2019-03-17:
感谢你的及时回复，问题已解决。此外我想问问是否有这个项目的相关文献或者这个项目的实现原理文档。谢谢你!

ruozhuxuan, 2019-03-17:
感谢你的及时回复，问题已经解决。此外我想问问是否有这个项目相关的论文文献或者项目的实现原理文档。谢谢您！

martin-ger, 2019-03-17:
I recently submitted a paper on this project, mainly focussing on the Automesh feature. Hope it will be published soon. Will mention it on this website as soon as possible.

ruozhuxuan, 2019-05-12:
您好，我在什么地方能看到关于这个项目的论文？