martin-ger/esp_wifi_repeater #286: ENC28J60 fails to start

bwmhamad, 2019-03-21:
I'm trying to make ENC28J60 work with NodeMCU. I followed the connection instructions and I'm using the 2N3905 transistor for decoupling GPIO15 along with 3.3k resistor. NodeMCU boots fine but from Serial output it seems to fail starting ENC28J60. Please note that I'm using _Nano V3 Ethernet Shield - ENC28J60 - RobotDyn_. Below serial output is looping:

>Repeater V2.2.6 starting
running rom 0
Config found and loaded
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
Starting enc28j60
ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34
rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  80 MHz
Booting rom 0.
⸮⸮⸮⸮o⸮{⸮⸮'|⸮dlld`c⸮⸮|{⸮d⸮g⸮⸮g⸮ l`⸮⸮s⸮l⸮d⸮d⸮H⸮H⸮⸮⸮

Thanks for help.

duile11, 2019-04-08:
change to Flash Speed: 160 MHz

ulsanjeewa, 2020-08-16:
Need a good power supply: the ENC28j60 needs about 160mA when active.

slowpoison4, 2021-03-15:
Two things to look out for:
1. Good power supply : I faced this problem as Martin points out, it takes 160mA to run the Module.
2. Check your connections. I faced the problem when D5 was connected to CLK on the board. It refused to boot up. I switched it to SCK on the Module and now it works. 
I however am not able to get Internet connectivity

martin-ger, 2021-03-15:
Ethernet should be your uplink? Do you use the STA-part of the ESP?

slowpoison4, 2021-03-15:
Hi Martin,
Im not very good with the terminology, But what i am basically doing is that I have an Active internet Lan -> Connect to ENC28j60->Esp8266 repeats this connection via MyAP-> Any IOT device can connect and have access to the internet. So far Ive got the code running but i am unable to get an active internet connection. Ive opened a separate issue regarding the same with a lot more details to help you help me diagnose the problem:
https://github.com/martin-ger/esp_wifi_repeater/issues/442