martin-ger/esp_wifi_repeater #294: Everything works except of repeating the packets

sabsik, 2019-04-02:
Hi Martin, thank you for the great project!

So far I flashed 2 NodeMCU ESP boards with your repeater. I disabled all remote configurations (both via Web and Telnet in the user-config.h) and only use serial port communicate with the boards.

After reboot I can ping the board from the Raspberry Pi connected on the sensor network (AP) and can ping that RPi back from the board using serial interface ping command. 

I also can ping ESP from the uplink device (another Raspberry Pi). I also can ping the uplink device back from the ESP using serial interface ping command. 

MQTT server on the uplink Raspberry Pi gets updates from ESP.

The only thing that does not work is I cannot ping the uplink RPi from the PRi connected on the sensor network. i.e ESP does not transfer the packets.

Note that out of two ECP boards I flashed, after a long struggle, one started working fully, but I did so many things that I cannot really recall in details now. The other one still does not transfer the packets.

How can I debug what is going on?

Here is output from the serial interface:

### After reset:
```
WiFi Repeater V2.2.6 starting

running rom 0
Config found and loaded
mode : sta(80:7d:3a:23:57:dc) + softAP(82:7d:3a:23:57:dc)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 11
cnt 
state: 5 -> 2 (2c0)
rm 0
disconnect from ssid pimon, reason 2
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 11
cnt 

connected with pimon, channel 7
dhcp client start...
connect to ssid pimon, bssid b8:27:eb:95:5b:f4, channel 7
add 1
aid 1
station: b8:27:eb:69:9a:0c join, AID = 1
station: b8:27:eb:69:9a:0c join, AID = 1
ip:192.168.4.68,mask:255.255.255.0,gw:192.168.4.1
ip:192.168.4.68,mask:255.255.255.0,gw:192.168.4.1,dns:192.168.4.1
MQTT: Connected
```

### show config:
```
Version V2.2.6 (build: Sun Mar 24 22:22:09 2019)
STA: SSID:pimon *******************
AP:  SSID:MyExtended1  PW:**************** IP:192.168.42.1/24 [NAT]
STA MAC: 80:7d:3a:23:57:dc
AP MAC:  82:7d:3a:23:57:dc
STA hostname: ESP_2357dc
Clock speed: 80
MQTT: enabled

```
### show acl
```
CMD>
Packets denied: 0 Packets allowed: 0
CMD>

```
### show stats
```
System uptime: 0:02:46
5 KiB in (46 packets)
1 KiB out (20 packets)
Power supply: 2.924 V
Phy mode: n
Free mem: 13080
STA IP: 192.168.4.68/24 GW: 192.168.4.1
STA RSSI: -78
1 Station connected to SoftAP
Station: b8:27:eb:69:9a:0c - 192.168.42.2

```
### show route
```
Routing table::
Network              Dest
192.168.42.0/24      ew1
192.168.4.0/24       ew0
127.0.0.0/8          lo0
default              192.168.4.1

```
### show dhcp
```
DHCP table:cp
b8:27:eb:69:9a:0c - 192.168.42.2 - 116

```
### show mqt:
```
MQTT client connected
MQTT host: 192.168.4.1
MQTT port: 1883
MQTT user: none
MQTT password: 
MQTT id: ESPRouter_2357dc
MQTT prefix: /WiFi/ESPRouter_2357dc/system
MQTT QoS: 0
MQTT command topic: /WiFi/ESPRouter_2357dc/command
MQTT gpio_out topic: /WiFi/ESPRouter_2357dc/switch
MQTT interval: 15 s
MQTT mask: ffff

```
All changes I made to the files are here:

### File user/user_config.h
```
009 #define		WIFI_SSID "pimon"
010 #define		WIFI_PASSWORD "***************"
011
012 #define		WIFI_AP_SSID "MyExtended1"
013 #define		WIFI_AP_PASSWORD "***************"
014
015 #define     WIFI_AP_OPEN 0 // This is my invention, could just put 0 in user/config_flash.c
...
069 // Define this to 1 if you want to have access to the config console via TCP.
070 // Ohterwise only local access via serial is possible
071 //
072 #define		REMOTE_CONFIG 0
073 #define		CONSOLE_SERVER_PORT 7788
074
075 //
076 // Define this to 1 if you want to have access to the config via Web.
077 //
078 #define		WEB_CONFIG 0
079 #define		WEB_CONFIG_PORT 1986
...
099 // Define this to 1 if you want to offer monitoring access to all transmitted data between the soft AP and all STAs.
100 // Packets are mirrored in pcap format to the given port.
115 // CAUTION: this might be a privacy issue!!!
102 //
103 #define		REMOTE_MONITORING 0

```
### File user/config_flash.c
```
057    config->auto_connect                = 1;
...
062    config->ap_open			= WIFI_AP_OPEN; // It is 0 from the user_config.h
...
089    IP4_ADDR(&config->network_addr, 192, 168, 42, 1);
...
120 #if MQTT_CLIENT
121    os_sprintf(config->mqtt_host,"%s", "192.168.4.1");

```
Thank you,
Sergei

martin-ger, 2019-04-02:
Did you test this with my provided binaries? This is the basic functionality - it usually works... ;-)

sabsik, 2019-04-03:
Sorry, my fault. I had to shut down eth0 on the client PRi to make ping working to the uplink RPi. everything works now. Absolutely great project!

closed 2019-04-03 by sabsik