martin-ger/esp_wifi_repeater #302: Trying to configure it breaks it

MatiHalperin, 2019-04-17:
When I try to update a setting from web interface or serial it ends unusable after reboot

`Repeater V2.2.9 starting

running rom 0
No config found, saving default in flash
Loading default configuration
Saving configuration
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(xx:xx:xx:xx:xx:xx) + softAP(xx:xx:xx:xx:xx:xx)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
add 1
aid 1
station: xx:xx:xx:xx:xx:xx join, AID = 1
station: xx:xx:xx:xx:xx:xx join, AID = 1
Saving configuration
Restarting ... 
vPortFree 515 line assert fail
del if0
station: xx:xx:xx:xx:xx:xx leave, AID = 1
rm 1
vPortFree 515 line assert fail
bcn 0
del if1
usl

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz

Rom 0 is bad.
Rom 1 is bad.
No good rom available.
user code done

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz

Rom 0 is bad.
Rom 1 is bad.
No good rom available.
user code done`

martin-ger, 2019-04-17:
Thanks for the report. Think it was a problem with the size of the code (hit config data). Should be fixed with the lastest update.

MatiHalperin, 2019-04-17:
Working now, but I have a new issue. Configured it and started moving it through the house to find a good place. It worked at first. But now it's in a loop where it restarts again and again. I flashed the firmware again, but it´s still loading the config, which appears to be corrupted or wrong

`Repeater V2.2.9 starting

running rom 0
Config found and loaded

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz

Booting rom 0.`

That´s showing again and again, that´s why I guess it´s rebooting

martin-ger, 2019-04-17:
Try "reset factory" (or flash something to address 0x61000. This will erase the config.)

MatiHalperin, 2019-04-17:
Working again, nvmd. Thanks for the quick fix!

closed 2019-04-17 by MatiHalperin