martin-ger/esp_wifi_repeater #305: possible to lock MAC addresses in mesh mode?

thrusty66, 2019-05-07:
Hi, is it possible to lock MAC addresses in mesh mode?
So that Handy´s (sell phones), tablets can not log on to the ESP. 
Or is there any other way to lock out certain devices?

martin-ger, 2019-05-07:
No, I am not aware of such an option. Either you know the WPA password and you can connect or you don't know and stay out...

thrusty66, 2019-05-07:
Da es mein Heimnetzwerk ist kenne ich das Passwort. 😉
Das bringt mich aber leider nicht weiter. Ich kann im Mesh keine unterschiedlichen Passwörter vergeben.

martin-ger, 2019-05-07:
If don't use the home access point as level 0 router, but configure the first level of ESPs manually, you can use a different password for all following Automesh nodes.

thrusty66, 2019-05-07:
I don't know that option.  But it sounds good for now.  Do you have a link where I can read it?

thrusty66, 2019-05-08:
I tried to change the password of the ESP in AutoMesh Mode. So I tried to use a different password than the one I assigned to the main network. Unfortunately the ESP does not take over this change.

martin-ger, 2019-05-08:
Switch off Automesh mode on the first level of ESPs. Use them as "normal" repeaters with a new SSID/Password on the AP side. Then configure the remaining ESP as Automesh for this new SSID.

bchabrier, 2019-05-08:
I had the same problem and fixed it with a slight change to the automesh mode: it now uses ap_ssid (and possibly ap_password) to find an uplink bssid.

In this mode, each ESP will either connect to the root SSID **or** to the closest ESP, thus optimizing the quality of the ESP network. However, the ESPs will be reachable through their own ssid (and possibly password).

I have create PR #306 in case you find this improvement useful.

thrusty66, 2019-05-08:
Thank you so much for your help. So far it works now. 
So this request can be closed.