martin-ger/esp_wifi_repeater #307: MQTT message from subnet to main network

thrusty66, 2019-05-08:
Now that my subnet and mesh are running, my next problem. 
How do I get mqtt messages from subnet to main network and back?

bchabrier, 2019-05-08:
You must have a mqtt server in your main network. For each ESP you must specify its IP in `mqtt_host`. Each ESP will connect to the server from its subnet. 
On their side, each ESP subscribes to the mqtt server, which means they receive published messages (especially those publish on topic `/WiFi/ESPRouter_XXX/command`).

Hope it clarifies!

thrusty66, 2019-05-08:
Sometimes it can also be quite simple 😁. Mosquitto has been running with me for a long time.

**I just didn't know that messages from the subnet were sent directly to the main network and upside down. Without making any further settings or installations.**

Now I can register all concerned Sonoff´s to their new subnet. 😃
Thank you very much for the quick help. 
This project is really helpful once you understand how it works. 😉

closed 2019-07-22 by martin-ger