martin-ger/esp_wifi_repeater #308: No response on NodeMCU after Flashing

NineNineNineIonoSpheres, 2019-05-09:
I have flashed my NodeMCU V.1.0 following this video: https://www.youtube.com/watch?v=5vGix-qOPK8 as well as the steps from https://github.com/martin-ger/esp_wifi_repeater#building-and-flashing 

However, when I hit the reset button on NodeMCU V.1.0, the LED light blinked just once and nothing happened, unlike in the youtube video above at 2:34 which keep on blinking.  I've checked with wifi analyzer app and there appears to be no wifi spreading from the NodeMCU.

Monitoring the Serial port, I got nothing in 115200 baud . On 74880 baud, however, I got some text:

```
20:28:38.229 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,2)
20:28:38.229 -> 
20:28:38.229 -> ets_main.c 
20:28:39.583 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,2)
20:28:39.583 -> 
20:28:39.583 -> ets_main.c 
20:28:43.515 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:28:43.515 -> 
20:28:43.515 -> ets_main.c 
20:30:17.087 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,4)
20:30:17.087 -> 
20:30:17.087 -> ets_main.c 
20:35:07.698 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,5)
20:35:07.698 -> 
20:35:07.698 -> ets_main.c 
20:35:33.999 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
20:35:33.999 -> 
20:35:33.999 -> ets_main.c 
20:35:39.624 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,2)
20:35:39.624 -> 
20:35:39.624 -> ets_main.c 
```

Not sure if this is related, but the number for boot mode seems to change from time to time...

I've also tried uploading an ESP8266 Wifi Access Point example program using Arduino, and I got wifi signal and could connect to the NodeMCU. 

So what should I do here?

martin-ger, 2019-05-10:
Try DIO mode for flashing, also make sure, the binaries are not corrupted (best is downloading the repo as zip and extracting it afterwards).

CylCeramic, 2019-07-27:
I had a similar problem with a NodeMcu v3 and also with a NodeMcu v2.
Everything seems to run smooth but "MyAp" never appeared in the list of wifi networks. Even downloading the complete repo as zip.
Reading once again the README.md, found this: "You can also interactively switch between two installed binaries.." so I decided to play with that.
this combination:

0x02000.bin @ 0x02000
0x82000.bin @ 0x82000

as shown in the captured, and the blue LED linked to D0 (GPIO16) is on and "MyAp" in the networks list.
[image]

MagTun, 2019-12-26:
I had the same problem (the wifi didn't appear). I tried several configuration but nothing worked including the one just above. 

The solution was to press the reset button ("RST") on the NodeMCU after the "download" is finished.  Even the official configuration worked (I just change the baud rate - I use the one indicated in the "esp download tool" command window)