martin-ger/esp_wifi_repeater #309: Unable to enable MQTT

angadsule, 2019-05-11:
Unable to enable MQTT using set mqtt_host

i have telnet to 10.24.1.1 7777 and then 
trying to set mqtt_host using
set mqtt_host "IP Address" i m getting error of invalid command

martin-ger, 2019-05-12:
Did you really try e.g.?
`set mqtt_host 192.168.0.2`

angadsule, 2019-05-13:
I tried it but there was problem but then I flashed the firmware again n it
was successful.

elfinharpy, 2019-05-16:
I'm having the same problem. I have set mqtt_host, mqtt_user and mqtt_password to the correct settings. I've also tried reflashing the firmware

"show mqtt" displays

MQTT client disconnected
MQTT host: 192.168.1.5
MQTT port: 1883
MQTT user: ######
MQTT password: ######
MQTT id: ESPRouter_0a3772
MQTT prefix: /WiFi/ESPRouter_0a3772/system
MQTT QoS: 0
MQTT command topic: /WiFi/ESPRouter_0a3772/command
MQTT gpio_out topic: /WiFi/ESPRouter_0a3772/switch
MQTT interval: 15 s
MQTT mask: ffff

bchabrier, 2019-05-20:
Can you post the output of `show config` ?

elfinharpy, 2019-05-21:
`show config` outputs:

Version V2.2.9 (build: Sun May 12 09:22:21 2019)
STA: SSID:##### PW:######
AP:  SSID:MyAP  PW:none [open] IP:192.168.4.1/24 [NAT]
STA MAC: 5c:cf:7f:0a:37:72
AP MAC:  5e:cf:7f:0a:37:72
STA hostname: ESP_0a3772
Network console access on port 7777 (mode 3)
Clock speed: 80
MQTT: disabled

martin-ger, 2019-05-21:
Did you do a "save" and "reset"?

elfinharpy, 2019-05-24:
No I didn't. That's made it work

Thank you

closed 2019-05-24 by martin-ger