martin-ger/esp_wifi_repeater #311: Porting WiFi Repeater to ESP32

mantatcheung, 2019-05-19:
Do you have the plan to port this to ESP32?  ESP32 is a newer and more power chip.

martin-ger, 2019-05-20:
As mentioned in the README there is a port of the basic NAT functionality to the ESP32 at https://github.com/jonask1337/esp-lwip . Currently I have no plans to port the complete SW. With the NAT port however, it should be farly easy to port any part of its functionality.

mantatcheung, 2019-05-20:
Thanks a lot!
With ESP8266 at 160MHz, I can achieve a good performance for web browsing and fair quality video streaming.  However, multiple users joined the repeater, the usability is reduced.  I'm intending to use ESP32 to get a higher performance.  One challenge that I faced now is installing the ESP32 SDK which requires Python supporting SSL.  Tried to install Python 3.7 with SSL a few times but failed.  I'll keep researching it.

mantatcheung, 2019-05-25:
For your update, I've setup the ESP32 with esp-idf and successfully make up the example that you mentioned.  When I flash the ESP32, I use make menuconfig to select 240MHz for the ESP32 chip.  The performance of which is much better than that of ESP8266.  It can go more than 10Mbps.

henkiejan1, 2019-06-12:
> For your update, I've setup the ESP32 with esp-idf and successfully make up the example that you mentioned. When I flash the ESP32, I use make menuconfig to select 240MHz for the ESP32 chip. The performance of which is much better than that of ESP8266. It can go more than 10Mbps.
> The next step I'd like to do is to port it to Arduino. Any tips for porting to Arduino?

Do you have a step by step instruction. Im really curious :)

martin-ger, 2019-06-12:
No step by step instructions, sorry. I have no experience with the ESP32 Arduino environment. What I did for the ESP8266 Arduino was to replace the installed lwip from the SDK against the port. This worked pretty straight forward: https://github.com/martin-ger/lwip_nat_arduino

conashin, 2019-08-29:
@mantatcheung 
Can you show us how to port wifi repeater into esp32?
I really want to know how?

martin-ger, 2019-08-29:
You can use this as a starting point: https://github.com/jonask1337/esp-lwip
The basic NAT works - it will need some kind of user interface. You can try to use my CLI code from user_main.c or try to start with a better and more complete web interface.

QGB, 2020-02-03:
@mantatcheung @conashin 
Can you share us your code and firmware binnay?

conashin, 2020-02-11:
@QGB I can't even understand all of the code😅

mantatcheung, 2020-02-15:
Sorry for not responding to your message earlier.  I was terribly busy recently.  However, I will upload my code somehow.

mahesh2000, 2020-02-16:
@martin-ger, any chance you could post the binary for the ESP32? i've had trouble installing the ESP IDF. thanks!

martin-ger, 2020-02-16:
Could do that easily, however, the current demo uses compile time constants 
for SSIDs and passwords. Thus is useless for arbitrary envirunments. First 
somebody should add at least a minimal config Option.

mahesh2000, 2020-02-16:
You weren't able to port that over from your ESP8266 implementation? From what I gather, your ESP32 version uses a different code base. Also, the ESP32 seems to come with mesh networking. Are you making that available? Thanks!

martin-ger, 2020-02-16:
No, was not able, as the system interface for the ESP32 is completly different from the ESP8266 NON_OS_SDK. But, hey, we provided you with the working core functionality! Could do new coding, if I were paid for - but I am not... 

Maybe I find some time to do more for the ESP32, but actually there is no real need for it: the cheaper ESP8266 does this job.

And ESP Mesh cannot be combined with the NAT feature, ESP Mesh isn't based on IP as far as I understood so far.

martin-ger, 2020-02-17:
Find a very basic version here: https://github.com/martin-ger/esp32_nat_router
It does the basic NAT and can be configured with serial console (Putty or GtkTerm with 115200 bps). Use the "set_sta" and the "set_ap" command to configure the WiFi settings. Enter the help command get a full list of all available commands.

closed 2020-02-17 by martin-ger

mantatcheung, 2020-03-01:
> For your update, I've setup the ESP32 with esp-idf and successfully make up the example that you mentioned. When I flash the ESP32, I use make menuconfig to select 240MHz for the ESP32 chip. The [...]

> @mantatcheung
> Can you show us how to port wifi repeater into esp32?
> I really want to know how?

I posted it to my repository with captive portal.  Document of which is not completed yet.

mahesh2000, 2020-03-01:
@mantatcheung, did you post a new version based on @https://github.com/martin-ger/esp32_nat_router?
> I posted it to my repository with captive portal. Document of which is not completed yet.

mantatcheung, 2020-03-02:
No, it is a version which I created on my own before @martin-ger posted his version.  The README is not completed and some comments in the code are not tidied up.  I'll complete it in the coming 2 weeks.