martin-ger/esp_wifi_repeater #316: Not working on ESP-01S with DOUT flash

czlowiek1, 2019-05-27:
After flashing firmware it prints errors on UART. Sketches from Arduino IDE work well after setting flash mode to DOUT.

martin-ger, 2019-05-28:
What does it print?

czlowiek1, 2019-05-29:
Now I can't check it, because I flashed NodeMCU, but it depended on tool, which I used to flashing. When I used esptool built in Arduino IDE, It printed something with ets_main.c at end, something simillar to:
` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1396, room 16 
tail 4
chksum 0xef
load 0x00000000, len 0, room 4 
tail 0
chksum 0xef
ho 12 tail 0 room 4
load 0x00000000, len 0, room 12 
tail 0
chksum 0xef
csum 0xef
csum err
ets_main.c `
When I used ESP8266 Download Tool, it printed errors all the time. It looked like numbers in registers.

martin-ger, 2019-05-29:
Did you check the sha1 checksums? Try to download the zip and extract the binaries from there. Direct download sometime corrupts them.

czlowiek1, 2019-05-30:
Finally it works after flashing using NodeMCU PyFlasher. Other flashing software does not work. I combined all .bin files into one using hex editor, because PyFlasher doesn't have offset settings.