martin-ger/esp_wifi_repeater #318: nat problem

fenglaijun, 2019-05-30:
i had build a net with 3 level mesh,and the number of esp-8266 are 60.finally,the wifi connection is build up success,but some of the esp-8266 client can't connect the first level's server.so this project can't hold big numbers mesh work?or do you make some limitation in code?

martin-ger, 2019-05-30:
There is no intentional limitation. The size of the NAT table is a limitation: currently there is a limit of 512 simultanious connections. This is typically not a severe limitation, as TCP entires are set free again as soon as the FIN packet is seen. UDP entires are held for 2 secs, when no packet is seen. A problem might occur if you really use many connections or if - for some reasons, TCP connections are not closed properly. TCP entries are held open for 1800 sec, i.e. half an hour.

Currently you cannot change the size of the NAT table (without soucre code changes in my esp_open_lwip), but you can change the connection timeouts. You might try to reduce it, e.g.:

`set tcp_timeout 60`

Does that change the behavior? 

In the next version I will add variable size of the table and some info on the current usage in the show commands.

What is also known: Due to the limitations of the ESP's SoftAP implementation, there is a maximum of 8 simultaniously connected stations. (https://github.com/martin-ger/esp_wifi_repeater#known-issues ).

martin-ger, 2019-05-30:
Just added 
`set max_nat value`

You might try larger values than 512.

fenglaijun, 2019-05-31:
thank you for your quick help!i will try this

closed 2019-07-22 by martin-ger