martin-ger/esp_wifi_repeater #320: Can't connect to Home router, STA mode

koudelapavel, 2019-06-02:
Hi,
very useful project, thank you Martin. But, I have small problem. If i flash the binaries to NodeMCU, all works fine, I can connect over repeater to Internet. After that, I tries to flash binaries to ESP-12F in my actually project. I used NodeMCU as programmer, can't find USB to Serial programmer. So, in this case, the serial monitor shows the message below continuosly, and the NAT doesn't work. Any ideas please?

`scandone
no "MY_HOME_NETWORK_SSID" found, reconnect after 1s
disconnect from ssid "MY_HOME_NETWORK_SSID", reason 201
reconnect
`

martin-ger, 2019-06-09:
For some reason it doesn't find the AP. Some RF issues?
```
REASON_UNSPECIFIED              = 1,
   REASON_AUTH_EXPIRE              = 2,
   REASON_AUTH_LEAVE               = 3,
   REASON_ASSOC_EXPIRE             = 4,
   REASON_ASSOC_TOOMANY            = 5,
   REASON_NOT_AUTHED               = 6,
   REASON_NOT_ASSOCED              = 7,
   REASON_ASSOC_LEAVE              = 8,
   REASON_ASSOC_NOT_AUTHED         = 9,
   REASON_DISASSOC_PWRCAP_BAD      = 10,  /* 11h */
   REASON_DISASSOC_SUPCHAN_BAD     = 11,  /* 11h */
   REASON_IE_INVALID               = 13,  /* 11i */
   REASON_MIC_FAILURE              = 14,  /* 11i */
   REASON_4WAY_HANDSHAKE_TIMEOUT   = 15,  /* 11i */
   REASON_GROUP_KEY_UPDATE_TIMEOUT = 16,  /* 11i */
   REASON_IE_IN_4WAY_DIFFERS       = 17,  /* 11i */
   REASON_GROUP_CIPHER_INVALID     = 18,  /* 11i */
   REASON_PAIRWISE_CIPHER_INVALID  = 19,  /* 11i */
   REASON_AKMP_INVALID             = 20,  /* 11i */
   REASON_UNSUPP_RSN_IE_VERSION    = 21,  /* 11i */
   REASON_INVALID_RSN_IE_CAP       = 22,  /* 11i */
   REASON_802_1X_AUTH_FAILED       = 23,  /* 11i */
   REASON_CIPHER_SUITE_REJECTED    = 24,  /* 11i */

   REASON_BEACON_TIMEOUT         = 200,
   REASON_NO_AP_FOUND              = 201,
   REASON_AUTH_FAIL            = 202,
   REASON_ASSOC_FAIL            = 203,
   REASON_HANDSHAKE_TIMEOUT      = 204,
```

MagTun, 2019-12-26:
I had a similar problem
```
no SFR_D555 found, reconnect after 1s
disconnect from ssid SFR_D7B0, reason 201
```
The box around the nodemcu was shielding the wifi signal.

Eletroholic, 2020-05-30:
Same issue .
What is the solution

Eletroholic, 2020-05-30:
> For some reason it doesn't find the AP. Some RF issues?
>
> ```
> REASON_UNSPECIFIED              = 1,
>    REASON_AUTH_EXPIRE              = 2,
>    REASON_AUTH_LEAVE               = 3, [...]

What is its solution .. kindly reply i m stuck

brandon-piner-aruba, 2021-07-02:
I get this too. Its very weird because it can scan for the network but then gives a 201

To connect use the command line and `disconnect` then `connect`

But this does not solve if the device restarts

TooMuchAir, 2022-03-22:
I had the same issue (error 201). I also discovered, that my ESP was trying to connect to a Wifi network that I had configured in a completely different project. 
It seems that the Wifi settings (SSID & PW) are saved and not overwritten when flashed.
For me the issue was resolved by using the ESP8266 Download Tool and erasing before flashing:

[image]

If anyone could confirm the resolution, perhaps @martin-ger can update the readme?