martin-ger/esp_wifi_repeater #321: Can't set a center Dns server to this repeater.

James0329, 2019-06-04:
Under tcpip config have a command call "set dns dns-addr:".
But when I use telnet to repeater try to set this command like 192.168.1.254 , the command result is 255.255.255.255  .
How can I set it like I want?
Should I use automesh mode?
Because some unspoken reason I have to setup a local dns server in LAN like dnsmasq to solve some special dns issue.
And I can't use any public dns server.
Anyone could help me?

martin-ger, 2019-06-04:
Cannot reproduce this behavior:
`set dns 9.9.9.9`
gives the correct setting for me.

What does your `show` tell you after setting the DNS IP?

Did you do a `save` before a reset?

James0329, 2019-06-05:
I copy your command into telnet program it work well.
I think I know what happen.
It cause by microsoft input method.
I didn't change this to english mode.So there have some symbol belong to my mother language .
Command screen can't understand symbol "。"
This symbol equal dot in my language.
This dot trigger 255.255.255.255 behavior.
Thank you for your help.

closed 2019-06-05 by James0329