martin-ger/esp_wifi_repeater #325: [Feature Request] STA Speed/Ping Test with LED blinking

caogia, 2019-06-23:
Hi, I installed the firmware and everything work fine. I also soldered and external antenna from and old router, but not really better.
Everytime I want to check the location of the ESP where it get highest speed from STA. So I would request a feature like LED blinking same as speed it does already connected to STA hotspot.
Would this possible?
Thanks

martin-ger, 2019-06-24:
I don't understand exactly the intended functionality. You want to see the RSSI encoded in blinking speed?

caogia, 2019-06-26:
sorry for my bad english.
  Yes, would be RSSI indicator. I mean LED will blink as the signal_strengh/internet_access indicator after connected to hotspot to locate where the best to install the esp(and maybe stop this function for some seconds)
  Also, I can't change phy_mode to n mode, it always in g mode

caogia, 2019-06-26:
would the NAT can automatic change the STA  if it has a better performance/ better ping? So we provide a list of hotspot, and set a ping level. In case of lower than ping, it will automatically change the STA?