martin-ger/esp_wifi_repeater #327: See devices connected in main router page

Muplex, 2019-06-28:
Ciao,
thank you for your work. I find it very usefull and I saved some money making myself a range extender with an Esp8266 - 01 that work very well.

My question is very simple but I did't find any reference in other posts.

Is it possible to see the sensors connected on the wifi-repeater (subnet 192.168.4.XXX) on the page of main router (with IP 192.168.178.XXX) ?? 
My objective is to monitor if the domotic sensor is connected to the net or not.

I tried to set the subnet same range as main but after that I had to re-flash the module 

Thanks in advance for your help.

Muplex, 2019-06-30:
Hello, I discover that the two sensors can send message to my domotic hub with the mqtt protocol without problems over the repeater to my home network. Thats Great. All works very well

The only thing I ask you to understand is if it's possible to ping the two devices 'cause in these casa I could know if they still connected to the net or they disconnect.

martin-ger, 2019-06-30:
Pinging them is not possible. What you can do is using the MQTT LWT mechanism to check, whether they are online:
- define two topics (one for each sensor), let's say "status/sensor1" and "status/sensor2"
- when a sensor connects, it publishes "online" on its topic as retained message
- it also registes "offline" on this topic as retained LWT
- as soon as the broker dtetects a connection loss, it will publish this message
- any client can check at any time the status of the sensors by subscribing to "status/sensor1", "status/sensor1" or "status/#"

hafened, 2019-07-25:
hello,how esp 8266 can be a wifi repeater and also burn a code in it.(in a sane esp).i have issue the program overwrite or crash,actually i update firmware for repeater and write source code in arduino IDE..thanks

martin-ger, 2019-07-25:
This is not an Arduino project. You cannot use this code in Arduino.

hafened, 2019-07-31:
> This is not an Arduino project. You cannot use this code in Arduino.

thanks,is it possible wifi repeater and a program e.g blinking led can burn in same esp?or any other alternate way?..thankyou