martin-ger/esp_wifi_repeater #330: Repeater disconnection problem

Autobot86, 2019-07-09:
Hello,
I'm using repeater firmware using 2 ESP8266 12E module. I have connected the 1st ESP with router.2nd ESP is connected with 1st ESP using 1ESPs SSID and password. 3rd ESP IS connected with 2nd ESP, using 2nd ESPs SSID and password. But there appears to be some problem while connecting 2nd esp with 3rd esp, How to solve to this issue. Because my project involves connecting ESPs together to overcome WIFI range issues.

martin-ger, 2019-07-09:
In general, I'm not aware of any problems when chaining ESPs. But perhaps you might try the automesh mode?

Autobot86, 2019-07-09:
Thanks for replay martin-ger,
I try automesh mode. when try to connection device getting reset as below

vPortFree 515 line assert fail
del if0
usl
16:23:29: 
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   16 Mbit
Flash Mode:   QIO
Flash Speed:  80 MHz

Booting rom 0.

martin-ger, 2019-07-09:
That's all? No more boot message?
If so, the firmware is somewhat corrupted.  If you are in doubt concerning the firmware binaries being corrupted, download the complete repo as zip and extract the binaries from that zip - this avoids HTTP-download problems (e.g. CR-LF conversions).

Autobot86, 2019-07-10:
Again I download firmware, flash tool and upload it. when 2 ESP node are nearest placed its working good. if we will increase the distance these are all the bellow problem arrised. when Device scanning find router its take continuous reset

 scandone
Found: 3,"demotest",-79,"20:34:f5:53:32:22",6, mesh level: 0
Using: 20:34:f5:53:32:22
Saving configuration
vPortFree 515 line assert fail
del if0
usl
12:42:14: 
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz

Booting rom 0.
„ã[03]ì[13]ÃgäÛsƒò'|ä[0C]$Œä$`[03][1C]#“[1B][02][04]Ÿ|[03]{’[03]$Ünà[04]ƒoâ[00][04]l`[03]Œâs’l„l[1B]Ä[0C][04][0C]lXH¨Hè•[19]¥Repeater V2.2.13 starting

running rom 0