martin-ger/esp_wifi_repeater #332: I2C sensor monitoring

vangalvin, 2019-07-11:
This may be a silly question, I have a project on that requires a bunch of sensors to collect data and fire it off to a central location at the same time as the ESP's are running as a meshed network. 
Ideally I would like to have the sensors connected directly to the ESP mesh router and have the data collected sent to an online database. Sensors are checked every 20 seconds or so.

I also need to be able to put the whole system in to a deep sleep mode then wake it up based on the time of an RTC to conserve power as all the units will be battery powered. however I also need to integrate a global broadcast that tells the system that if the sensor data is being sent over the network the nodes do not enter deep sleep until the data has been sent.

Would this project have the ability to support such functions?

martin-ger, 2019-07-12:
Some thoughts about that:
- The esp_wifi_repeater has no I2C IO capability.
- A sleep cycle of 20s (needing some ms for measurement and storing data (flash or RTC mem?)) is still quite power consuming (for a battery). The ESP is not really a low power chip.
- A global data collection phase can be implemented with the esp_wifi_repeater: it can be controlled via MQTT. So a data collector can wait until it received all data (probably also via MQTT) and then send out a "sleep <s>" on a common control topic. esp_wifi_repeater has some delay before actually going to sleep, so the message will be propagated before all fell asleep.

vangalvin, 2019-07-16:
Thanks, Yes it looks like I am going to have to investigate other options for this one. Sadly due to the remote location of this system changing batteries is not going to be an option and as there is snow and ice solar is not going to work well. wind speeds also get up to 200kph so a wind generator is not going to last long. 

The one bonus I do have is that the ground temperature close to where the sensor is located is about 60 degrees C and the ambient air temperature is -20 degrees C so I am looking at the possibility of using a peltier to charge the system. Couple that with a Solar panel that may or may not end up with snow/ice on it and I may be able to keep this thing running.

peterloron, 2020-03-04:
@vangalvin You would probably be better off looking at LoRa radios. They are designed specifically for long range, very low power, intermittent connectivity.