martin-ger/esp_wifi_repeater #334: "station leave" not recognized by esp_wifi_repeater

BLWG, 2019-07-20:
Hi, thx for the great software, but I think I have a small problem.
Using the precompiled libraries there s the prob that "station leave" is not recognized everytime by repeater. Neither on serial nor on mqtt it s then shown. Please see sample log below:

add 1
aid 1
station: e4:2b:34:55:9e:f0 join, AID = 1
station: e4:2b:34:55:9e:f0 join, AID = 1
station: e4:2b:34:55:9e:f0 leave, AID = 1
rm 1
station: e4:2b:34:55:9e:f0 leave, AID = 1
add 1
aid 1
station: e4:2b:34:55:9e:f0 join, AID = 1
station: e4:2b:34:55:9e:f0 join, AID = 1
station: e4:2b:34:55:9e:f0 leave, AID = 1
rm 1
station: e4:2b:34:55:9e:f0 leave, AID = 1
add 1
aid 1
station: e4:2b:34:55:9e:f0 join, AID = 1
station: e4:2b:34:55:9e:f0 join, AID = 1

//till here everythings s ok, now I disconnect station: no "leave"message apears
//reconnect station again:

err already associed!
station: e4:2b:34:55:9e:f0 leave, AID = 1
rm 1
station: e4:2b:34:55:9e:f0 leave, AID = 1

that means: many times it works as expected, sometimes doesnt. Tried it with different stations.
"leave" in that case is shown at stations reconnect followed by "join" immediatly.

thx in advance, Bernd

martin-ger, 2019-07-22:
Here I am relying on the events given by the esp wifi lib. I have actually no idea, why it is missing the leave. Obviously, it is still thinking, that it is connected: the "err already associed!" is not printed by my software...

BLWG, 2019-07-22:
Even it doesnt help me, thx for your fast answere. Why didn t answere you to the issue-list to let other people know it s not from your lib? I think if I ll close the issue other people could be worried,

thx a lot, Bernd

Mit freundlichen Grüßen / Regards
ppa Bernd Lehnert
Abt. V
Vertrieb / Sales department
Service
___________________________________________

RL-Messtechnik GmbH
Tel.: +49 2104 173539
Mob.: +49 151 15259479
Fax.: +49 2104 173541
Lehnert@RL-Messtechnik.de

BLWG, 2019-07-22:
Excuse me, cause I m using the precompiled bins, I can not see the includes. What lib do you think is the problem? The ESP8266WiFi.h ?
thx Bernd

Mit freundlichen Grüßen / Regards
 ppa Bernd Lehnert
Abt. V
Vertrieb / Sales department
Service
___________________________________________

RL-Messtechnik GmbH
Tel.: +49 2104 173539
Mob.: +49 151 15259479
Fax.: +49 2104 173541
Lehnert@RL-Messtechnik.de

martin-ger, 2019-07-22:
Yes, the events come from the closed source WiFi driver.

How do you enforce the disconnect? By switching off the STA?

BLWG, 2019-07-22:
The station is another esp8266 or my iphone. To enforce the disconnect at esp-staion I simply disconnect the voltage. At iphone by disabling wifi. Result with both stations the same. Many times it works, sometimes not. Its really a pity because your software does exactly what I was looking for. Your repeater inside garage connected to the homewifi, the station-esp connected to ignition of the car. Dooropen when station established, doorclose after "station leave". 
Dooropen always (seems) to work, doorclose not.
BTW: Does GPIO set/read works with the precompiled bins?
Thx Bernd

Mit freundlichen Grüßen / Regards
ppa Bernd Lehnert
Abt. V
Vertrieb / Sales department
Service
___________________________________________

RL-Messtechnik GmbH
Tel.: +49 2104 173539
Mob.: +49 151 15259479
Fax.: +49 2104 173541
Lehnert@RL-Messtechnik.de

martin-ger, 2019-07-22:
As there is no heardbeat from a STA to an AP, the AP cannot realize, that a STA suddenly dies. A STA misses the beacons, but an AP doesn't expect a STA to send any periodic message.

What about equipping the car with an ESP AP (witj no further functionality) and building a simple AP as MQTT sensor that scans for its SSID?

Yes, GPIO option is compiled in the binaries.

BLWG, 2019-07-22:
Hey Martin, thx a lot for the explaination. So it seems not to be a bug. I ll think about your idea. Thx again fot great work. Can I copy your statement to the issue list to close that issue?

Mit freundlichen Grüßen / Regards
ppa Bernd Lehnert
Abt. V
Vertrieb / Sales department
Service
___________________________________________

RL-Messtechnik GmbH
Tel.: +49 2104 173539
Mob.: +49 151 15259479
Fax.: +49 2104 173541
Lehnert@RL-Messtechnik.de

martin-ger, 2019-07-22:
It is already in the issue desc.

Cheers,
Martin

closed 2019-07-22 by martin-ger