martin-ger/esp_wifi_repeater #340: Can't receive a Webcam stream

superfino, 2019-08-16:
Hi,
I use this esp8266-Repeater to extend my Wifi range for a external webcam. Nothing spezial with the configuration. I can ping the cam but can't receive the stream on my main network. On the AP's network everything works. Tried portmapping without success. Any idea why this happens or is a spezial setup required? Thanks in advance for any hint.

martin-ger, 2019-08-17:
You almost certainly need one or more portmaps. Do you know the protocol/ports that the webcam uses?

superfino, 2019-08-17:
Yes, the cam uses TCP and Port 80 for the web interface and 8081 for the MJPEG stream direct.

martin-ger, 2019-08-17:
Then you should use:

portmap add TCP 8080 192.168.4.2 80
portmap add TCP 8081 192.168.4.2 8081

Now you should be able to access the cam at the ESPs address at port 8080.

superfino, 2019-08-17:
O.K. thanks Martin,
it was my faulty reasoning. As mentioned, I dealed already with this portmaps but I've tried to reach my cam on their 192.168.4.2:8080 IP address instead of 192.168.0.17:8080, the interface to my router network.  Thanks again for your help and this great piece of code.
Jo

closed 2019-08-17 by superfino