martin-ger/esp_wifi_repeater #341: ESP don't connect to uplink AP #279 reopen

kock-rueil, 2019-08-22:
Idem issue 279: ESP does not connect to AP.  I made tests with one nodemcu and two different AP.

martin-ger, 2019-08-22:
Maybe the AP doesn't accept IEEE 802.11g clients? In STATIONAP mode the ESP can only work in g mode, not in n.

kock-rueil, 2019-08-25:
After a couple of tests: connexion is KO in WPA and OK in WEP security and/or without security. Wifi n or g, both work correctly. In which part of the firmware is the security managed ? Thanks for your help Martin.

martin-ger, 2019-08-25:
It is configured here https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/user_main.c#L4042

But I am not aware of any general problem here. In fact, this is standard code and it works just fine in my setup. Maybe the pw ist too short? 8 chars minimum.

kock-rueil, 2019-08-26:
Changing password including two blank caracters with a new password without blank and it works good in WPA. ESP8266webserver library doesn't have this issue. What do you think about that ?