martin-ger/esp_wifi_repeater #342: Can't connect to wifi networks with spaces & special characters

xmagickx, 2019-08-31:


xmagickx, 2019-08-31:
I have read all the previous issues of space character... Now in the web interface of ESP repeater, I want to enter "Shyam's wifi"... But even after using %20 or \ as suggested by you @martin-gerr, it fails to connect & shows the ssid as "Shyam" only.. Please help me out...

duts33, 2019-09-06:
Такая же фигня. Не подключается к сети название которой состоит из нескольких слов и разделено пробелом.

duts33, 2019-09-06:
> I have read all the previous issues of space character... Now in the web interface of ESP repeater, I want to enter "Shyam's wifi"... But even after using %20 or \ as suggested by you @martin-gerr, it fails to connect & shows the ssid as "Shyam" only.. Please help me out...
Use Telnet Commands. Thus, we managed to get to work with an access point that has spaces in the name. Through the web interface, too, did not work.

PanagLtd, 2019-12-19:
> Use Telnet Commands. Thus, we managed to get to work with an access point that has spaces in the name. Through the web interface, too, did not work.

Telnet command is not working as well. The ssid ends at the first space character.

lordpt, 2021-04-18:
I use %20 to set space not in web page form but in url address and work for me.