martin-ger/esp_wifi_repeater #346: Wemos D1 mini (esp8266) board dosen't work,,,

opensupportfactory, 2019-09-23:
I have "Wemos D1 mini" board , one of esp8266 boards. I downloaded the firmware to the board then it didn't work any more and when I tried to see the debug logs from the problematic board using teraterm program, the Wemos D1 mini didn't output any logs on teraterm, serial program.

Maybe it seems that the wifi repeater's firmware didn't support the wemos D1 mini.
Is it right?

martin-ger, 2019-09-23:
Usually, it does support it. Have you tried any other firmware with the same flashing parameters? It so and this works, perhaps try to clear flash before flashing the repeater binaries.

opensupportfactory, 2019-09-23:
I have "nodemcu V2", another board and when I test it on nodemcu board, it worked well as a wifi repeater then the problem occured only if on wemos D1 mini. When I tried to download other firmware running as a webserver with aduino IDE on wemos D1 mini board, it worked well as a webserver.

But when I tried a wifi repeater's firmware using ESP8266 Download Tool on wemos D1 mini with same flashing parameters as nodemcu V2 board's, the same problem happened like above.

opensupportfactory, 2019-09-23:
When I looked into the issues with "wemos D1 mini", I found like belows:

"Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector. Flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1)."

So the my problem is like that after reboot, I see only garbage on the serial and when I try to re-initialize that sector as above with "esp_init_data_default.bin", I couldn't find "esp_init_data_default.bin" under that path, currently.

What do I have to do currently?

martin-ger, 2019-09-23:
The problem has been solved in later releases, as it flashes its owe "esp_init_data_default.bin" on boot.

If you still think, it could be this problem, you might want to go back and try an older FW, let's say this one: https://github.com/martin-ger/esp_wifi_repeater/tree/c74cdcf3f4783beb7a8b6c2ac5a753a8abb55d13

opensupportfactory, 2019-09-23:
It doesn't seem that the problem with "esp_init_data_default.bin" differed from my issue because the problem still happened when I also tried the older FW (https://github.com/martin-ger/esp_wifi_repeater/tree/c74cdcf3f4783beb7a8b6c2ac5a753a8abb55d13) on Wemos D1 mini, board.

But when I tried to debug my issue, Wemos D1 mini, board only outputed some garbage data for a second after reboot and I coudn't see any log from the board.

I'm confused what problem is with the board now.
Any advices?

HotCodeworker, 2019-09-25:
I also have the above problem with Wemos D1 mini, board (https://studymake.tistory.com/597).
I see the garbage data via serial program after reboot. but it works well with example appliactions in arduino IDE. What is the problem?

zorruno, 2019-11-27:
These are the D1s I have https://www.aliexpress.com/item/32649549788.html they have worked fine (flashed last couple of days).  If you are using the ESP8266 Download Tool - I found that first using 'erase' helps when reflashing.  Earlier versions of this tool didn't have that option.

jarno83, 2020-01-21:
Same here, I didn't get it work with Wemos D1 Mini PRO. :(

Emmpunkt, 2020-01-28:
Same here on two different boards. Bought two new ones, the same :(
The garbage data is at 74880 baud and says:
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
ets_main.c

martin-ger, 2020-01-28:
"rst cause:2" means reset pin (https://www.esp8266.com/viewtopic.php?p=2096#p2112 )
This should be no software issue. Power issue?

Emmpunkt, 2020-01-28:
No, i found the solution ;)
The firmware was corrupted.
I dowloaded the whole zip (as you suggested in your manual) and it works now on 3 devices.
Wer lesen kann ist klar im Vorteil (german phrase)

jarno83, 2020-01-29:
> No, i found the solution ;)
> The firmware was corrupted.
> I dowloaded the whole zip (as you suggested in your manual) and it works now on 3 devices.
> Wer lesen kann ist klar im Vorteil (german phrase)

Thanks, Yeah now works. I flashed all 3 files with same offsets as file names.