martin-ger/esp_wifi_repeater #347: Feature request: captive portals auto login

xaxeu, 2019-09-24:
If you want to have ESP repeater connected to a public wifi, you need it to be able to auto login the captive portal using a username/password.

Is this feature possible?

martin-ger, 2019-09-25:
To do this automatically is quite difficult, as most public hotspots have different input mask and login schemes (only PW, username/PW, just okay click,...)
The easiest is: the first repeater user logs in automatically, afterwards all others ride on this ticket.

xaxeu, 2019-09-25:
I tried to connect one repeater to a public wifi, but I'm not redirected to any portal and I don't even have connectivity to any dns server.

What do you mean with "first repeater user logs in automatically"? Or how do you achieve this?

Thanks.

martin-ger, 2019-09-25:
When you connect the esp_wifi_repeater to a hotspot and then connect a client to the repeater, you should be redirected to the auth page. Once you are there, the manual interaction with the auth page should enable the esp_wifi_repeater's MAC address for full internet access and so all of it's clients as they are transparent behind the NAT (IP and MAC). At least for me this worked in several hotels...

For Windows clients, the mechanism works like this: https://blog.superuser.com/2011/05/16/windows-7-network-awareness/ For Apple and Google similar mechanisms with different addresses are implemented.

When an  esp_wifi_repeater connects to a captive portal hotspot, it should get an address of a valid DNS server. This server is afterwards also forwarded to any client of the  esp_wifi_repeater. You might check this using "nslookup" on the  esp_wifi_repeater's console.