martin-ger/esp_wifi_repeater #350: Issue with ESP01 - No Access point available

juancgalvez, 2019-10-01:
Hi Martin.

Thank you for your excellent software.

I have it in a Wemos D1 Mini and it works great with some IoT devices that I have far from my main router. Absolutelly stable.

Recently I found some old ESP-01 with 512KB memory and flashed esp_wifi_repeater into them. None worked. I can get access it via serial console, but there is no AP available to connect to.

This is the erasing and flashing result:

```
d:\Electronica Micro Controladores\ESP8266\esptool-master>python esptool.py --port COM6 erase_flash
esptool.py v2.5.1-dev
Serial port COM6
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:9b:89:b1
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 1.3s
Hard resetting via RTS pin...

d:\Electronica Micro Controladores\ESP8266\esptool-master>python esptool.py --port COM6 write_flash -fs 512KB -ff 80m -fm dio 0x00000 0x00000.bin 0x02000 0x02000.bin
esptool.py v2.5.1-dev
Serial port COM6
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:9b:89:XX
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash params set to 0x020f
Compressed 1968 bytes to 1410...
Wrote 1968 bytes (1410 compressed) at 0x00000000 in 0.1 seconds (effective 113.2 kbit/s)...
Hash of data verified.
Compressed 386752 bytes to 264269...
Wrote 386752 bytes (264269 compressed) at 0x00002000 in 23.7 seconds (effective 130.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```
This is the flash ID information:

```d:\Electronica Micro Controladores\ESP8266\esptool-master>python esptool.py --port COM6 flash_id
esptool.py v2.5.1-dev
Serial port COM6
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 18:fe:34:9b:89:XX
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4013
Detected flash size: 512KB
Hard resetting via RTS pin...
```
After flashing I reset it and get this:

```
Starting Web Config Server on port 80
mode : sta(18:fe:34:9b:89:XX) + softAP(1a:fe:34:9b:89:XX)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
```

If via serial console I send the command **show config**, I get this:

```
show config
Version V2.2.13 (build: Sun Sep  1 11:20:03 2019)
STA: SSID:ssid PW:password [AutoConnect:0]
AP:  SSID:MyAP  PW:none [open] IP:192.168.4.1/24 [NAT]
STA MAC: 18:fe:34:9b:89:XX
AP MAC:  1a:fe:34:9b:89:XX
STA hostname: ESP_9b89XX
Network console access on port 7777 (mode 3)
Clock speed: 80
MQTT: disabled
CMD>
```

But, Is mentioned before, no MyAP access point is available to connect to.

If I manually configure ssid, password, device static IP, netmask, Gateway then save configuration and reset I get this:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   DIO
Flash Speed:  80 MHz

Booting rom 0.
????N??{??o|?$??l B??|r?$?N??n?Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(18:fe:34:9b:89:XX) + softAP(1a:fe:34:9b:89:XX)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid WDR3600, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid WDR3600, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid WDR3600, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid WDR3600, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid WDR3600, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (0)
disconnect from ssid WDR3600, reason 202
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid WDR3600, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (0)
disconnect from ssid WDR3600, reason 202
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (0)
disconnect from ssid WDR3600, reason 202
reconnect
```
I am using the latest firmware up to date downloaded from github.

**What could I be doing wrong?**

martin-ger, 2019-10-01:
Very strange - if you go back to an older fw, let's say beginning of this year. Better?

juancgalvez, 2019-10-01:
Martin.

Thank you for your fast reply.

I tried several esp_wifi_repeater versions:

```
2019-01-20
2018-11-08 (Keeps resetting)
2018-11-26
2018-06-24 (Uses 0x00000 and 0x10000)
```
And none of them worked.

Thinking it could be an issue with de ESP-01 I flashed **DNSServer -> CaptivePortal** demo for ESP8266 in Arduino IDE and it worked flawlessly. My computer sees it's AP, can connect and get a response from it.

juancgalvez, 2019-10-01:
Sorry for closing the thread. I thing I did something wrong. I am reopening it.

juancgalvez, 2019-10-01:
Could the problem be related to the fact that the device has only 512KB?

juancgalvez, 2019-10-04:
Hi Martin.

May I expect some additional reply from you?

Thank you.

martin-ger, 2019-10-04:
Short reply: I don't know. Please remember: this is my spare time project and it is free to anybody, so don't expect 24/7 support! Also, thousands of users have installed this successfully in the last years. So it is know to work in many environments. And it is known to work with ESP-01 - even the latest version runs for me. 

Think my ESP-01 has 1 MB, so 512 might be the issue. You might try to go back to this version, it was before rboot. This should definitly fit into 512 KB.

https://github.com/martin-ger/esp_wifi_repeater/tree/f3c299011254d8f37e0f4b875e149689a715fb18

juancgalvez, 2019-10-05:
Hi Martin.

Excuse me. I didn't want to  bother you. I just tought you were looking at something. I understand you are doing it for free to help people like me.

I tried the versión you told me, and it showed the AP and I was able to connect to it a configure. But, once configured it kept disconnecting from the router almost every 5 seconds. Disconnect reason was 34.

I thing I am going to give up with these ESP-01 512KB devices.

Thank you for your great firmware and help.

Txemin, 2019-10-05:
Hi Juanc,

I´ve been  stuck at the same point  for 1 month, and I still am. I' working with with NodeMCU, it has 4 MB. Now I'm busy with another matters but still thiniking about it. I've even thought that the problem might be my router, but I don't know what should be the next step to take. I'll try giving it a fixed IP adress, Just to try something. 
It's a shame that we can't take advantage of so much good work.
I haven't quit yet.

Gisbert1, 2019-10-08:
Hi Txemin,
I flashed the latest firmware to a Sonoff Dual Vers. 1 and to an ESP8266-01 with 1MB size, both worked right from start as designed. Absolutely no issues with flashing, getting started and using the devices. No restarts so far, but only 2.5 days after installation.

Best Regards
Gisbert

Txemin, 2019-10-20:
Thank you very much. At the end I didn't need to do it. After changing power unit, and learning port forwarding it's working.