martin-ger/esp_wifi_repeater #352: traffic accounting per-IP basis

andrewerix, 2019-10-06:
Can you help me to figure out some of the ways it is possible to perform traffic accounting? I want to do accounting on a per-IP basis using esp-wifi-repeater. I just need the total traffic usage per-IP in megabytes. I have looked into the traffic shaping and the firewall source code and also the ESP core documentation, API reference and also the SDK source code but couldn't find any related content.

martin-ger, 2019-10-06:
It is done in the functions my_input_ap() and my_output_ap() here: https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/user_main.c#L488 . Look for Bytes_in and Bytes_out.

These two funktions are hooked into the lwip stack (between the driver and the TCP/IP) here: https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/user_main.c#L3941

Then you can see every packet in and out at the IP interface and you can do the accounting.

In the esp_wifi_repeater you can get these values via MQTT.

andrewerix, 2019-10-08:
That's great. I saw that daily limit has been implemented too. How to set the daily quota limit? 

I don't know how to publish (to MQTT broker) the total Bytes_in and Bytes_out for different users connected to the softAP. Should I hook sta_ip and sta_mac to patch_netif as well? Can you help me with the IP Interface by giving an example?

martin-ger, 2019-10-09:
- set daily_limit limit_in_KB: defined a max. amount of kilobytes that can be transferred by STAs per day (0 = no limit, default)

Look into this sample: https://github.com/martin-ger/lwip_nat_arduino/blob/master/CaptiveHotspot.ino

Here I do a similar thing. Once you have the IP header of a packet on the AP interface, you can identify the client and make per-client (IP) stats. This is not done in the esp_wifi_repeater.