martin-ger/esp_wifi_repeater #353: Access Ip adress from device connected to the esp

Chaosflori25, 2019-10-06:
Hey Guys,
following problem: I wanted to connect an Ip camera to my esp, but access it over the main network. Is it possible, that I can access the camera connected to the repeater from the main network, and if it is possible, how?
Thanks a lot 4 any kind of help

Gisbert1, 2019-10-08:
Hi,

Portmapping is the key word.
Look at the Readme document, there you will find the right explanation.
I have access from my main network to the IP addresses from the ESP repeater/router network.
Secondly have installed a static route in my main router to the ESP network. I'm not sure whether this is really necessary, but it seems also not to harm.

Best Regards
Gisbert

Chaosflori25, 2019-10-09:
Thanks a lot

closed 2019-10-09 by Chaosflori25