martin-ger/esp_wifi_repeater #354: ESP Wifi Repeater on a Sonoff Dual

Gisbert1, 2019-10-06:
Hi Martin,

I flashed your firmware to a Sonoff Dual, and basically all worked fine.
With the Sonoff Dual I have 2 relays, both are not controlled via GPIO but with serial commands like these:
void switch_relay(byte b) {
  if (0<=b && b<=2) {
    Serial.write(0xA0);
    Serial.write(0x04);
    Serial.write(b);
    Serial.write(0xA1);
    Serial.flush();
  }
}

In the setup this needs to be present:
  Serial.begin(19200);

The relays are the switched on/on by:
switch_relay(0); // both relays are off; possible values: 0, 1, 2

Is there a way to implement serial commands to your firmware and send serial commands?
Can you describe how it might work?

Best Regards
Gisbert

Gisbert1, 2019-11-14:
Dear Martin,

you have explained to me (in another channel) how to compile your code (Ubuntu in a VM Virtual Box, SDK, make, ...). Basically I was successful and could produce the binaries.

So far so good, but now I'm at a point where I have no chance to go forward as I cannot do programming. I don't ask for a solution, but only for your estimation if such a code in C++ (Arduino) or similiar code in C have a chance to work:

> Serial.begin(19200);
void switch_relay(byte b) {
if (0<=b && b<=2) {
Serial.write(0xA0);
Serial.write(0x04);
Serial.write(b);
Serial.write(0xA1);
Serial.flush();
}
}

The variable b needs to be send to the ESP, probably via MQTT.
Is it possible to modify one of the files in the mqtt folder?

Best Regards
Gisbert

martin-ger, 2019-11-14:
The esp_wifi_repeater uses the serial interface as console to communicate with the user. And it is used for flashing. I don't know exactly, how the Sonoff Dual works. Does it ignore any other output on the serial? If so, then it might be easy:

The serial baud rate should be set to 9600 here: https://github.com/martin-ger/esp_wifi_repeater/blob/f6c02fb63d1777cac087b68616689c0b6341f609/user/user_main.c#L4249

And a new mqtt handler has to be installed here: https://github.com/martin-ger/esp_wifi_repeater/blob/f6c02fb63d1777cac087b68616689c0b6341f609/user/user_main.c#L225

Something like:
```
#define SONOFF_TOPIC "sonoff_topic/mytopic"

    if (topic_len == os_strlen(SONOFF_TOPIC ) && os_strncmp(topic, SONOFF_TOPIC , topic_len) == 0 && data_len >= 1)
    {
        uint8_t buf[4];
        uint8_t b = data[0] - '0';
        if (b < 0 || b > 2)
              return;

        buf[0] = 0xA0;
        buf[1] = 0x04;
        buf[2] = b;
        buf[3] = 0xA1;
        uart0_tx_buffer(buf, sizeof(buf));
        return;
    }
```

Gisbert1, 2019-11-14:
Dear Martin,

Gisbert1, 2019-11-14:
Dear Martin,

I put the code below line 225:

> #define SONOFF_TOPIC "sonoff_topic/mytopic"

    if (topic_len == os_strlen(SONOFF_TOPIC ) && os_strncmp(topic, SONOFF_TOPIC , topic_len) == 0 && data_len >= 1)
    {
        uint8_t buf[4];
        uint8_t b = data[0] - '0';
        if (b < 0 || b > 2)
              return;

        buf[0] = 0xA0;
        buf[1] = 0x04;
        buf[2] = b;
        buf[3] = 0xA1;
        uart0_tx_buffer(buf, sizeof(buf));
        return;
    }

I got this error:

> AR build/app_app.a
build/app_app.a(user_main.o): In function `tcp_monitor_sent_cb':
/home/gisbert/Downloads/esp_wifi_repeater-master/user/user_main.c:368: undefined reference to `uart0_tx_buffer'
build/app_app.a(user_main.o): In function `to_scan':
/home/gisbert/Downloads/esp_wifi_repeater-master/user/user_main.c:4227: undefined reference to `uart0_tx_buffer'
collect2: error: ld returned 1 exit status
Makefile:132: recipe for target 'firmware/0x02000.bin' failed
make: *** [firmware/0x02000.bin] Error 1

I'm try to find out what it means. In the next 10 days I'm travelling - I will continue working on it when I'm back at home.

Best Regards
Gisbert

jackkum, 2019-11-15:
@Gisbert1 
Try replace 
```
uart0_tx_buffer(buf, sizeof(buf))
``` 
[to](https://github.com/martin-ger/esp_wifi_repeater/blob/master/driver/new_uart.c#L227) 
```
UART_Send(0, buf, sizeof(buf))
```

I found only definition of [this](https://github.com/martin-ger/esp_wifi_repeater/blob/07f9da0b9c7d3121e95a64f215ac684d2582cceb/include/driver/uart.h#L206) method, but no implementation, so linker throw error.