martin-ger/esp_wifi_repeater #357: Multi-SSID ?

12368w, 2019-10-22:
esp_mesh сеть из 20 esp8266
мне необходимо менять имя (ssid) каждой отдельной ячейки в mesh сети, отталкиваясь от состояния датчика или управляющих команд,  отдельно от других ячеек - но в общей mesh сети ?

пример ssid1 - R15H100, ssid2 - R30H70, ssid3 - M30R40 ,,,, ssid,n ---

внешнее устройство реагирует на название (ssid) точки - без подключения
точки отдельно меняют свои названия - но находятся в одной mesh сети

есть возможность реализовать Multi-SSID ?

прошу любые мысли - как это сделать?

martin-ger, 2019-10-23:
Multi-SSID is not supported by the Espressif WiFi driver and it probably nevver will be. I don't think there is any chance to do this as it has to be done in the Layer 2 which is not open.

12368w, 2019-11-02:
другие мысли?
мы делаем игру с применением esp8266 в виде артефактов
ESP8266 вещает wifi с именем эффекта - телефон не подключаясь понимает эффект из названия
мне нужно менять эффект - имя ssid wifi
и желательно все esp8266 в одной esp-mesh объединить

25 ~ 30  esp8266
любые идеи?
менять имя на ходу от датчика, на время, в ячейке?