martin-ger/esp_wifi_repeater #358: How about adding a web page authentication？

callmebg, 2019-10-24:
Hello! I am a Chinese student. I am very interested in this project. In reality, some wifi require web authentication before they can be used. Can you add this feature? For example, add three input boxes, authentication URL, username, password, and post or get. Looking forward to your reply.

martin-ger, 2019-10-26:
This can be done, however the build-in web server is rather basic so it would be a major change. Also, there is currently no mechanism to filter single users (block them or let them pass).

Have a look at this: https://github.com/martin-ger/lwip_nat_arduino/blob/master/CaptiveHotspot.ino
This should do most of the things you expect. Just add a password-logic there.

niclasfredrik, 2019-10-26:
I have done a concept for this subject. External captive server. I would be
happy to contribute. Cheers

martin-ger, 2019-10-26:
Feel free to contribute!

callmebg, 2019-10-26:
> This can be done, however the build-in web server is rather basic so it would be a major change. Also, there is currently no mechanism to filter single users (block them or let them pass).
>
> Have a [...]

Your code is great! However, I have no way to correctly import the lwip you provided.
I am sorry that my bad English made my statement unclear. In fact, I want to solve such a problem. In my university, we covered campus wifi without passwords. In other words, when we connect to wifi, we can connect successfully without a password. However, if we want to connect to the Internet instead of the LAN on campus, we need to enter our student number and password on a web page.
My bad university can only allow one device to connect to wifi. So, I want to connect my campus wifi with esp8266 and make sure I'm connected to the Internet by entering our authentication URL, student ID and password. Then, share this network.
In other words, can you add authentication pages in sta mode instead of ap mode?

martin-ger, 2019-10-26:
Think, what you want to do can easily be done with the existing esp_wifi_repeater:
- configure ap_ssid and ap_password
- connect your first device to the esp_wifi_repeater
- connect the esp_wifi_repeater to the campus wifi
- try to connect to the internet from the connected device
- the login page should show up
- login on the connected device
- you should be able to reach the internet
- now connect other devices to the esp_wifi_repeater
- they should be able to connect to the internet without prior login, as the campus wifi sees only the MAC of the esp_wifi_repeater and thinks, this is one computer.

callmebg, 2019-10-26:
good job！Thank you！------------------ 原始邮件 ------------------

  
 
 
 
 
 
 
 
 
  
&mdash;

agoktugaydin, 2022-12-17:
> 

hello, 
first of all, thank you very much for providing this project.

but I have a problem with the same situation. In order to access the internet connection in my dormitory, I must connect to the internet network and then enter my personal information on the login page. but unfortunately, esp cannot connect to the public dormitory network. because after entering the ssid and password information, when I connect to the access point I created from esp, the login page does not appear.

also, when I access the access point with the IP address again, the password field is not empty. so I think the esp could not connect to the network because there is an old password instead of an empty password box. while there was no problem when I connected to another clear network, it bothered me a lot to encounter this problem when I connected to the dormitory network. What can I do to get past this? do you have any idea about this?

Alacritous, 2022-12-17:
Campus internet admins are like reddit mods. They take their jobs way too seriously. You're probably being blocked by the network admins because they're very paranoid about multiple devices connected to their networks through chained connections.  You may be running afoul of their network rules by trying to connect to their network through a repeater.

agoktugaydin, 2022-12-17:
> Campus internet admins are like reddit mods. They take their jobs way too seriously. You're probably being blocked by the network admins because they're very paranoid about multiple devices [...]

I understand. I guess, since the dormitory I'm staying in is a state dormitory, they overdo the security measures.