martin-ger/esp_wifi_repeater #360: Developer Documentation : How Modify /Contribute

gpism, 2019-11-07:
Hi This project is amazing and I want to contribute but I noticed there is no developer documentation also this project doesn't not seem to use platformIO or Arduino   ...so no clue how to start ....is possible to provide documentation ...like which module does what and how compile /build .........even better if its possible to convert to platformio project

martin-ger, 2019-11-07:
Build instructions can be found here: https://github.com/martin-ger/esp_wifi_repeater#building-and-flashing

gpism, 2019-11-07:
Thanks Martin ...my bad must have missed it....is there developer
documentation on various modules ....does from start from user module? ..

niclasfredrik, 2019-12-16:
Dear Martin Ger,
I write this to contribute to the project. In a somewhat unique way.

My intention is to provide free WiFi to restaurant guests in Latin America.
Tourists would know the locations of the restaurants from an Android app,
or from a web page. A demo of the initial concept can be seen at
https://bonored.com/demo or check out the video at
https://bonored.com/demo/img/Bonored_concept_en.mp4

Once the concept is used in the hotels, it will be easy to launch the free
wifi concept.

I know that the ESP32 is a more powerful unit than the ESP8266. ESP32 could
provide a speed of 10 Mbit, which is perfect for this application.

I have had 3 different programmers who had the task to make up a working
WiFi repeatar, they all failed. I would be happy to pay 1000 Euro for
assistance to make a repeater on the ESP32 platform, having an external
captive portal functionality. (All server functionality is specified and
coded.)

Who can assist me? Please advice.
Cheers
Pensamiento Factory SAS
Niclas Arnberger, CEO

martin-ger, 2019-12-16:
Niclas,

the esp_wifi_repeater has become actually a feature monster and I guess, you won't need most of the functionality. The basic repeater functionality for the ESP32 can be found in this project: https://github.com/jonask1337/esp-lwip . What is missing is the captive portal functionality and probably some management functionality.

The basic idea of a (local MAC-based) captive portal (again on the ESP8266) can be found here: https://github.com/martin-ger/lwip_nat_arduino/blob/master/CaptiveHotspot.ino. This can be ported straight forward to the ESP32. What is still missing is the connection to an external portal an especially (if needed) a working authentication method.

What do you want to achieve? Only access to your side (the captive portal) or general internet access after some form of authentication?

Cheers,
Martin

niclasfredrik, 2019-12-16:
Hi Martin, thanks for your prompt reply.
I know how to code, and I could learn how to handle the ESP32 path of your
project. But I rather focus on my parts, web development of the map and
tour projects, and Android development.
The functionality I desire is a simple wifi repeater, no bells and
whistles, just a wifi repeater. The SSID should be BONORED, no password. It
will connect to a wifi network. However, it should include code for
automatic update of the firmware.
At some point in time, I want a captive portal functionality. The intended
functionality is that a user log in by inserting his email. The system will
track if the user is authorized. If so, the user will access Internet, if
not, the user will be redirected to another web page.

Who can help me?

Cheers

martin-ger, 2019-12-17:
With the example from https://github.com/jonask1337/esp-lwip you can start right away with:

> a simple wifi repeater, no bells and whistles, just a wifi repeater. The SSID should be BONORED, no password. It will connect to a wifi network. 

Config of the uplink wifi network (SSID, PASSWORD) is static here, but for an initial test even this counld be fine.

OTA is something that has to be added, but another sample like this should help:
https://github.com/espressif/esp-idf/tree/master/examples/system/ota/simple_ota_example

Captive portal is a little bit more complicated and needs some more specification, but the basic framework is given in the example above.

Perhaps any of the followers here on github is willing to do that? Of course, I could do this as well, but I am currently heavily busy with my main profession and I cannot promise timely delivery nor more than best-effort support...

niclasfredrik, 2019-12-17:
Dear Martin, Thanks a lot for a good description. I will will try.
Please feel free to suggest any person that could do the future development.
Best Regards
Niclas Arnberger

HamzaHajeir, 2020-06-02:
@niclasfredrik Did you try out wifi-repeater with config portal on esp32 ?