martin-ger/esp_wifi_repeater #362: Request a simple WiFi router with 8266 module principle!

JackMing0, 2019-11-10:
It was great to see the blogger's project!
Request the help of the blogger! If I want to use 8266 module to implement a simple WiFi router! You don't need anything else! What am I supposed to do? Request a simple WiFi router with 8266 module principle!
In this! Respect to bloggers!

martin-ger, 2019-11-10:
Look for this repo for a simple and easy to understand basic repeater based on the same technique: https://github.com/martin-ger/lwip_nat_arduino

JackMing0, 2019-11-10:
First of all, thank you for your reply!
I read your advice!
It depends on the "lwip" library used to implement a WiFi router
Unfortunately, my current working platform is not Arduino development platform!
You can't use this feature under the Arduino development platform!
I rebuilt this project based on the development platform of non-os SDK provided by ESP8266!
I also saw the blogger introduce the "lwip" library header in the user_main.c file
Such as:
#include "lwip/ip.h"
#include "lwip/netif.h"
#include "lwip/dns.h"
#include "lwip/lwip_napt.h"
#include "lwip/ip_route.h"
#include "lwip/app/dhcpserver.h"
#include "lwip/app/espconn.h"
#include "lwip/app/espconn_tcp.h"

If I want to implement 8266 WiFi router function! Are the above libraries necessary?
Ask blogger to give directions!
Which libraries are needed?

If you can! Also hope that bloggers simply explain! 8266 WiFi routing implementation principle!
And under your current project! User_main. c file! Specific implementation of WiFi routing code! Where is it? Such words also facilitate understanding and learning!

JackMing0, 2019-11-11:
Thanks for your reply!
Unfortunately, my development platform is not arduino
My development was based on ESP8266 NONOS SDK 2.2

May I ask, the technical principle used in this project is NAT address forwarding data?
Developed 8266WiFi router to use the header file is that several!
Please point it out! Thank you very much!
Such as:
#include "lwip/ip.h"
#include "lwip/netif.h"
#include "lwip/dns.h"
#include "lwip/lwip_napt.h"
#include "lwip/ip_route.h"
#include "lwip/app/dhcpserver.h"
#include "lwip/app/espconn.h"
#include "lwip/app/espconn_tcp.h"
#include "lwip/app/encdhcpserver.h"
#include "lwip/app/ping.h"

Is there any API documentation about how to use it?

martin-ger, 2019-11-11:
For ESP8266 NONOS SDK use this lib: https://github.com/martin-ger/esp-open-lwip
API is documented here: https://github.com/martin-ger/esp-open-lwip/blob/enc_polling/include/lwip/lwip_napt.h
Installation is described here: https://github.com/martin-ger/esp_wifi_repeater#building-and-flashing

> Only if you don't want to use the precompiled library, checkout the sources from https://github.com/martin-ger/esp-open-lwip . Use it to replace the directory "esp-open-lwip" in the esp-open-sdk tree. "make clean" in the esp_open_lwip dir and once again a "make" in the upper esp_open_sdk directory. This will compile a liblwip_open.a that contains the NAT-features.

JackMing0, 2019-11-11:
First of all, thank you for your reply!
Blogger mentioned above! I've seen it before!
It doesn't really help me!
First of all, I am not very clear about the code steps to implement WiFi routing!
There are no relevant demo examples in your project!
This instance exists in the user_main.c file of the current project!
Too much code! Very difficult in the analysis process!

I tried:
I will https: //github.com/martin-ger/esp-open-lwip library
Introduced into ESP8266 NONOS SDK project! Header file in lwip library!
Some variable or method is not defined!

After many failed attempts! I've decided to give up!
For the project to achieve 8266 routing this function is very good!

Question:
This project should be able to get a bigger promotion!
It is a pity! Lack of documentation support! There are no instances of a single function being implemented!

Advice:
Add functional demonstration examples and documentation support! Try to make the project modular! Simplify the explanation! Easy to understand! It is recommended on the project to offer a full ESP8266 NONOS and put the features under the current project in that directory! Solve some file missing variable or method problem!

martin-ger, 2019-11-11:
Yes, you are right - this project has become a feature monster and its internal structure isn't documented well. It is grown over time and you can only get an idea of the basic idea and it's structure by going back in time in the commit history.

But exactly to help others to build their own ideas around the NAT routing feature i released the https://github.com/martin-ger/lwip_nat_arduino as today there is no good reason any more to use the NONOS SDK. Even Espressif mainly supports either Arduino or FreeRTOS.

If you want to understand it, look at: https://github.com/martin-ger/lwip_nat_arduino/blob/master/WiFiNatRouter.ino - less than 70 lines and it is all done.

JackMing0, 2019-11-13:
Thank you for your comments!

I transferred the project to the Arduino development platform for operation

Operation:
The following operations are in strict accordance with the blogger:
 https://github.com/martin-ger/lwip_nat_arduino  project installation instructions!

Question:
The following problems occurred during compilation:
Thank you for your comments!

I transferred the project to the Arduino development platform for operation

Operation:
The following operations are in strict accordance with the blogger: https://github.com/martin-ger/lwip_nat_arduino project installation instructions!
[image]

Question:
The following problems occurred during compilation:

正在编译项目...
make -C "C:\\Users\\mac\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip/src" install "TOOLS_PATH=\"C:\\Users\\mac\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-\""
使用 1.0  版本的库 ESP8266WiFi 在文件夹： C:\Users\mac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi 
exec: "make": executable file not found in %PATH%
为开发板 Generic ESP8266 Module 编译时出错。

The operation process：
        
[image]
[image]
[image]
[image]

JackMing0, 2019-11-13:
Ask for your help。。。。。

closed 2019-11-18 by JackMing0