martin-ger/esp_wifi_repeater #364: NAT table "full" And "LmacRxBlk:1"!

JackMing0, 2019-11-19:
NAT table full" and "LmacRxBlk:1" problem!

Excuse me
The above question! Concrete solution! What!

martin-ger, 2019-11-19:
NAT talbe full: larger NAT table, shorter timeouts with:

- set max_nat no_of_entries: sets the size of the NAPT table (default 512)
- set tcp_timeout secs: sets the NAPT timeout for TCP connections (0=default (1800 secs))
- set udp_timeout secs: sets the NAPT timeout for UDP connections (0=default (2 secs))

LmacRxBlk:1: no more buffers for send receive = too much load for the small ESP:

- set speed [80|160]: sets the CPU clock frequency (default 80 Mhz)

might help a little bit, but basically you should reduce the network load.

JackMing0, 2019-11-19:
Thanks for your reply!

NAT table full problem:
Do you mean max_nat no_of_entries?
Ip_napt_init (uint16_t max_nat, uint8_t max_portmap);

Should I set NAPT table to be big or small?

LmacRxBlk problem:
Setting speed [80 | 160] : setting CPU clock frequency (the default is 80 Mhz)
You mean high frequency or low frequency! Mine is the default 80MHz

martin-ger, 2019-11-19:
Yes, max_nat, NAPT table should be larger, frequency 160 MHz.

JackMing0, 2019-11-19:
Is doubling the size of the NAPT table defined feasible
Such as:
#define IP_NAPT_MAX 1024
#define IP_PORTMAP_MAX 64

martin-ger, 2019-11-19:
Yes

JackMing0, 2019-11-19:
New problems:
After adjusting the size of the NAPT table, and CPU frequency! When 8266 is connected to the router! About two to three seconds to restart, loop wireless restart