martin-ger/esp_wifi_repeater #367: MyAp not showing and blue LED is constantly open

SteliosLL, 2019-11-27:
When i first flashed my esp8266, i opened the wifi connections and i could see "MyAp". After i configured it i couldn't see the connection again. I tried everything. Flashed it using different settings, different binaries etc. but nothing worked. The esp8266 is not problematic because i tested some other things on it and it worked just fine. Can someone help me fix the problem?

eldraco, 2020-01-11:
I had the same issue. First flash with esptool.py when ok. I connected to MyAP, I configured it and connect to my wifi. I changed the name of the Wifi and worked. I connect with telnet and worked. I captured packets with monitor mode and a remote wireshark using netcat like in the video. 
Every time I used the monitor mode the board died. But after a reset it kept working.
And then suddenly after the last reset with the button the board does not show the AP.

I reflashed the original bin file that worked before, but still does not show the original AP. I waited. Nothing.

I also tried to flash a basic blink program and it worked fine. 

thanks

huydangquoc, 2020-05-16:
Same here, would like to add a voice to pop up this issue.
I'm able to interact w/ the board via serial monitor (run `show`, `help` and set few random things), even change the ap_ssid. It's just my devices (phone, laptop, table) never find MyAP (or anything else that I set).
Pretty sure the board works well as I used it for my other projects as well.

GMahmoudi, 2022-07-29:
I have the same problem. MyAP doesnt appear at the first start of ESP8266 after programming. I tested 5 modules (all ESP-01), only one of them works. Although all modules have successfully passed multiple test examples (blinking, web server, web client ...).
Using a serial monitor the module gives the following:
=====================================================
running rom 0
Config found and loaded
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(4c:11:ae:03:2b:1d) + softAP(4e:11:ae:03:2b:1d)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100

=====================================================

Note: For some modules MyAP does appear, but it disappear after making some settings (AP name, password...). In this case the serial monitor shows the following:
==============================================================
WiFi Repeater V2.2.13 starting

running rom 0
Config found and loaded
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(4c:11:ae:03:2b:2c) + softAP(4e:11:ae:03:2b:2c)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid Tenda_wifi, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
disconnect from ssid Tenda_wifi, reason 2
scandone
state: 0 -> 2 (b0)
state: 2 -> 0 (2)
reconnect
.......
================================================
Reason 2 means REASON_AUTH_EXPIRE. I saw also reason 15 (REASON_4WAY_HANDSHAKE_TIMEOUT) and 201 (REASON_NO_AP_FOUND). see [https://www.espressif.com/sites/default/files/documentation/2c-esp8266_non_os_sdk_api_reference_en.pdf](url)
@SteliosLL @eldraco @huydangquoc @
I'm using ESP8266 Download Tool V3.4.1 
thanks!

GMahmoudi, 2022-07-29:
According to martin's reply https://github.com/martin-ger/esp_wifi_repeater/issues/384#issuecomment-593151443
"Do you have the PUYA flash chips? I think they will fail, as the current flash write routines do not work there."

I noticed the following:
ESP-01 with flash memory Puya P25Q80H 8A1F39D doesn't work (I have 4 modules of them!)
ESP-01 with flash memory Puya P25Q80H 8B1F88D works (I have one module of this type :( )