martin-ger/esp_wifi_repeater #368: Different gateway and IP creates problem

thahseenkmuhammed, 2019-12-13:
Hi Martin,
I done repeater using your code to extend WiFi of my home router with details as Gateway IP: 192.168.1.1, Subnet : 255.255.255.0, DNS: 192.168.1.1. In our IoT project, server is connected to router with reserved dynamic IP 192.168.1.51 and other nodes are connected with static IP assigned, say 192.168.1.10. I am using repeater for one node (say static IP 192.168.1.10), but my issue is when connecting this node to repeater, it provides network details as Gateway IP: 192.168.4.1, Subnet : 255.255.255.0, DNS: 192.168.1.1, dynamic IP say 192.168.4.2. so server cannot able to connect with 192.168.1.10 because repeater makes it as say 192.168.4.2. how can make 192.168.4.2 same as 192.168.1.10. How to route it.

martin-ger, 2019-12-13:
Set a static IP for the repeater's STA interface:
```
set ip 192.168.1.10
set 255.255.255.0
set gw 192.168.1.1
```
and a static DNS for your repeated network
```
set dns 192.168.1.1
```
You might want to configure a static IP in the 192.168.4.0/24 subnet for your node, let's say 192.168.4.2.100 (default also for the first dynamic IP is the 192.168.4.2, but a static IP avoids problems, when it get's another IP for some reason).

Now define a portmap for the node (example for port TCP 80)
```
portmap add TCP 80 192.168.4.100 80
```
Do this for any port you want to connect to.
Save this in the repeater's config.

Now you can reach your node's port 80 from the main network as 192.168.1.10:80.

thahseenkmuhammed, 2019-12-17:
Thank you Martin. I followed your instructions. But unfortunately my problem doesn't get solved. Please see the illustration of my issue which I was discussed with you.
[image]

Then I changed the network of Repeater then I get as like below.
[image]

I didn't get communication between Server and Node. Connecting mobile phone/laptop to access internet through repeater is working well, only issue with my IoT project. What I can do to make communication between Server which is connected to Router and Node ESP which is connected to ESP-Repeater. Hoping to hear from you.

martin-ger, 2019-12-17:
The lower config cannot work, because the REPEATER's subnet and the ROUTER's subnet must be different (the repeater is in fact a NAT ROUTER, i.e. it routes between different subnets).

The upper config in general is fine, but I don't unterstand the link between the ROUTER and the NODE(ESP). There no such direct connectivity. If you establish the
`portmap add TCP 8080 192.168.4.2 80`
on the REPEATER, you can reach the NODE's port 80 as 192.168.1.x at port 8080 (or whatever port you specify in the portmap statement, CAUTION the REPEATER has it's own port 80). x is the outside address of the router (probably 10?).

thahseenkmuhammed, 2019-12-18:
There is no need for connection between NODE(ESP) and Router when Repeater is used. I have given the route between Router and Node along with Repeater to show the difference of network details happening. My aim is to boost the connectivity between router and Node using repeater so that no direct connection will exist between the two later, only through repeater they will communicate. But Server is always connected to Router which can communicate with Node through Router-Repeater-Node path. Illustration updated below just for your information.
[image]

HamzaHajeir, 2020-06-24:
I'm looking for the same concept. I've IoT Devices At backyard, They host webserver to control/communicate with them. 

And I'm running mosquitto MQTT broker in my computer (static IP 192.168.1.20).

The issue : I can't reach devices by their IP. 
Neither the node can reach my MQTT server. 

Am I missing something ?

HamzaHajeir, 2020-06-26:
Update : I've used portmap as @martin-ger  suggested. Now I can reach nodes web-servers using 8080, 8081 .. external port, tied internally to respective node. (make sure you save dhcp so node IP doesn't change).