martin-ger/esp_wifi_repeater #370: Feature request: SSIDs with blanks, Passwords with special characters

treki1, 2019-12-26:
Is it possible to implement this?
ATM is this not working.

martin-ger, 2019-12-26:
This is how it it intenden to be:

On the web interface: If you want to enter non-ASCII or special characters in the web interface you have to use HTTP-style hex encoding like "My%20AccessPoint". This will result in a string "My AccessPoint". With this hex encoding you can enter any byte value you like, except for 0 (for C-internal reasons).

On the command line: if you want to enter non-ASCII or special characters you can use HTTP-style hex encoding (e.g. "My%20AccessPoint") or, only on the CLI, as shortcut C-style quotes with backslash (e.g. "My\ AccessPoint"). Both methods will result in a string "My AccessPoint".

What do you think is not working?

treki1, 2019-12-26:
Ah ok. I dont know before.
Thx

closed 2019-12-26 by treki1