martin-ger/esp_wifi_repeater #374: Problem when change gpio_out topic

bobo3366, 2020-01-28:
Hi
Good work....
I have write the firmware on my Wemos D1 R2 and ESP8266-01.

I have a problem when change gpio_out topic (i think the command is "mqtt_gpio_out_topic")
the respons is  : "Invalid Command".

And another help......
On one of my projects I inserted an IR on the GPIO-0 of my ESP8266-01.
How can I manage with this firmware ....?
Thank you

martin-ger, 2020-01-28:
The "gpio_out" is a compile time feature that is disabled in the pre-build binaries. If you want to set the GPIOs, the best way would be to publish a "gpio x set high/low" command to the command_topic.

I don't completly understand you second question. What do you need? Interrupt handling on GPIO 0? This requires additional code and is not supported be the current firmware.

bobo3366, 2020-01-29:
Thank you for your answer.
My second question was about a simple circuit consisting of an ES8266-01 and a proximity sensor.
The proximity sensor is connected with the gpio0 of the ESP8266 and sends data via MQTT to an external server.
Is it possible to configure it with the current firmware (esp_wifi_repeater) or do changes have to be made to the code?
Thanks as always

[image]

martin-ger, 2020-01-29:
To so that, you really have to enable the USER_GPIO_IN option for gpio0 in user_config.h and recompile the code. Then you will get an MQTT Message on input change.

bobo3366, 2020-01-29:
I'm sorry
I have serious problems with the compilation.
1 - bash problem (5.0.3) solved by modifying the line in the "crosstool-NG / configure.ac" file by forcing the recognition of my version.
Now I keep having problems with the comedy "make STANDALONE = y".
<-> start
cp -f 1000-mforce-l32.patch crosstool-NG / local-patches / gcc / 4.8.5 /
make -C crosstool-NG -f ../Makefile _toolchain
make [1]: entry in the "/ home / roberto / esp-open-sdk / crosstool-NG" directory
./ct-ng xtensa-lx106-elf
make [2]: entry in the "/ home / roberto / esp-open-sdk / crosstool-NG" directory
make [2]: -c: Command not found
make [2]: -c: Command not found
make [2]: -c: Command not found
make [2]: *** No rules to generate the "xtensa-lx106-elf" target. Stop.
make [2]: exit from the directory "/ home / roberto / esp-open-sdk / crosstool-NG"
make [1]: *** [../Makefile:135: _toolchain] Error 2
make [1]: exit from the directory "/ home / roberto / esp-open-sdk / crosstool-NG"
make: *** [Makefile: 131: crosstool-NG / .built] Error 2

<-> end

martin-ger, 2020-01-29:
You are in Windows? You will need a 'make' command. Cygwin?

bobo3366, 2020-01-29:
No, I have lubuntu virtual-machine

martin-ger, 2020-01-29:
You should Install the 'make' command.

bobo3366, 2020-01-30:
I already have.
I have install esp-idf complete with all software.

bobo3366, 2020-01-31:
Ok, I have install on Ubuntu (Docker) server anc compilation is OK.
Now I have other query....
I have 1 value on GPIO 0 and when IR qorking I have 0 and returning on 1.

Where is the solution for have 0 normal mode and 1 when IR working ?.

Thanks

martin-ger, 2020-01-31:
As I used 'gpio_pin_intr_state_set(GPIO_ID_PIN(USER_GPIO_IN), GPIO_PIN_INTR_ANYEDGE);' it shoud send an MQTT message on any state change of the GPIO pin.

https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/user_main.c#L4307

bobo3366, 2020-02-01:
Hi
[Solved] as already described on your documentation (https://github.com/martin-ger/esp_wifi_repeater) in the "GPIO Config" paragraph it is possible to set:
"gpio [0-16] trigger [0-16] [monostable_NO | monostable_NC | bistable_NO | bistable_NC]"
In my case it was enough to configure the gpio to "monostable_NO", in this way I have the value 0 and if it intervenes IR the value goes to 1.
Thank you

bobo3366, 2020-02-01:
Sorry I have omit one information:
gpio 0 mode in
gpio 1 mod out
gpio 0 trigger 1 monostable_NO
....

result:
GPIO 0: in, triggers GPIO 1 as a monostable normally open                                                               GPIO 1: out