martin-ger/esp_wifi_repeater #377: balena OS on raspberry pi can't connect thru esp_wifi_repeater

mahesh2000, 2020-02-02:
hi, i have a raspberry pi zero (W) running *balena OS* connecting to the internet via a linksys wifi router that's setup as a softAP. if the esp_wifi_repeater replaces that router (as it's doing the same job), the raspberry pi can no longer connect to the internet. any idea what the issue is? i can see its IP address when i do `show stats`. the standard raspi zero release, buster, does not have this problem. i'll probably have to ask balena, but i thought i'd ask here first. thanks!

mahesh2000, 2020-02-03:
i believe balena OS uses port 22222 for ssh. not it should make a difference. is there a way to track all communication coming from this raspi? perhaps the secret is hidden somewhere in there.

martin-ger, 2020-02-03:
You can use the monitoring feature of the esp_wifi_repeater to monitor any traffic coming from the clients. 

But before that: what exactly want to connect? Into the balena client or outgoing. In inbound you will need a port forward.

mahesh2000, 2020-02-03:
i want the balena os running on the raspberry pi to connect outgoing. they use a reverse proxy so i can dial back in. the problem is more fundamental. i'm not able to connect to it using ssh on the same network. i'll try monitoring balena os using a usb terminal if i can enable it. i've asked them what logging i should do on their end. it'll be good to fix this issue. i'll try the monitoring on esp_wifi_repeater as well.

mahesh2000, 2020-02-20:
hi, the raspberry pi running balena OS is accessible using your [ESP32 NAT router](https://github.com/martin-ger/esp32_nat_router). thanks!!!

closed 2020-02-20 by mahesh2000