martin-ger/esp_wifi_repeater #378: Printer parameter problem

Gowenna, 2020-02-08:
Hello Martin;

I cannot configure the ESP to use the printer.
Here are the parameters:

My Internet Box has IP: 192.168.1.254
When I am connected on my Internet Box I see the ESP: 192.168.1.31
When I am connected to my Box I can ping the ESP: 192.168.1.31 it works.

When I connect to the ESP with my Smartphone, everything works, I can use the internet with 6Mo Upload and 6Mo Donwload.

Parcontre, when my computer is connected to my Internet Box: 192.168.1.254 and my Smartphone is connected to the ESP: 192.168.1.31, I can't manage to make a PING on my Smartphone which has the IP 192.168.4.2, and this is the same with my Printer I cannot ping.

My Internet box: 192.168.1.254
ESP: 192.168.1.31
Smartphone: 192.168.4.2

I did this:

portmap add TCP 8081 192.168.4.2 8081

In fact I don't know if he puts 8081, I don't really know what to put.

Thank you for your, and thank you very much for what you are doing.

Steve.

Gowenna, 2020-02-08:
I add the information of my printer which is connected on the ESP:

Printer model: Epson Stylus Office BX305
Network Type: Wireless LAN
TCP / IP configuration: Automatic
IP address: 192.168.4.3
Subnet mask: 255.255.255.0
Default gateway: 192.168.4.1

martin-ger, 2020-02-09:
You cannot reach a device connected to the ESP (address 192.168.4.x) from your other WLAN unless you have a port forward set. 

Pinging from the ESP net to a device in the other WLAN should work, but not the other way round.

To make a device reachable, you will need a port forward like the one you described in the first posting. Of cause it is important to faorward the right port(s) to the right internal address. I'm not shure, but I would try HTTP(S): 80 and 443, and XMPP:  5222.

Gowenna, 2020-02-10:
You mean he makes me face this: portmap add TCP 443 192.168.4.2 443
or : portmap add 80 443 192.168.4.2 80 ?

But after having this in the ESP, do I have to do something on the router (My Internet box)?

Thanks Martin.

martin-ger, 2020-02-10:
I would suggest:

```
portmap add TCP 80 192.168.4.2 80
portmap add TCP 443 192.168.4.2 443
portmap add TCP  5222 192.168.4.2  5222
```

This means all request to these ports to the external address of the ESP will be forwarded to the port of the client 192.168.4.2 (assuming this is the printer). I.e. to the 192.168.1.x network it will look like the ESP is the printer.

I hope these ports are enough. If you find out, it uses any other port: add it to the list.

Perhaps you have to enter the IP of the ESP (as printer) manually somewhere in the config as a discovery broadcast will not work.

Gowenna, 2020-02-11:
Hello Martin;

I tried, I didn't succeed.

Question: can I put several ports on the same ip?
Question: do I have to restart the ESP after adding an ip + port like the portmap add?

Here is some console information if that helps.

CMD>show
Version V2.2.13 (build: Sun Sep  1 11:20:03 2019)
STA: SSID:Freebox-******* PW:***************
AP:  SSID:Freebox-ZZ*****  PW:******* IP:192.168.4.0/24 [NAT]
STA MAC: *****************
AP MAC:  *****************
STA hostname: ESP_*******
Network console access on port 7777 (mode 3)
Clock speed: 80
MQTT: disabled
Portmap: TCP: 192.168.1.31:4238 -> 192.168.4.2:4238
Portmap: TCP: 192.168.1.31:4443 -> 192.168.4.2:4443
Portmap: TCP: 192.168.1.31:80 -> 192.168.4.2:80
Portmap: TCP: 192.168.1.31:443 -> 192.168.4.2:443

================================

CMD>show stats
System uptime: 0:26:36
2 KiB in (32 packets)
0 KiB out (5 packets)
Power supply: 2.909 V
Phy mode: n
Free mem: 47136
STA IP: 192.168.1.31/24 GW: 192.168.1.254
STA RSSI: -55
1 Station connected to SoftAP

==================================

CMD>show dhcp
DHCP table:
00:17:af:d1:d2:10 - 192.168.4.2 - 99974

====================================

Thanks again Martin for your help.

Gowenna, 2020-02-11:
Do you think I should also open ports on my internet router?

Gowenna, 2020-02-13:
Hello Martin;

I resumed everything from the beginning, I deleted the ESP8266 and I reinstalled esp_wifi_repeater.

My ESP: ESP8266 Lolin NodeMcu v3

I give you the exact imformations of the configuration

[image]

[image]

CMD>show
Version V2.2.13 (build: Sun Sep  1 11:20:03 2019)
STA: SSID:Freebox-A***** PW:******************
AP:  SSID:Freebox-Z*****  PW:*************** IP:192.168.4.0/24 [NAT]
STA MAC: *****************
AP MAC:  ******************
STA hostname: ESP_1a385b
Network console access on port 7777 (mode 3)
Clock speed: 80
MQTT: disabled
CMD>

======================================

CMD>show stats
System uptime: 0:21:27
6650 KiB in (7472 packets)
5608 KiB out (6528 packets)
Power supply: 2.908 V
Phy mode: n
Free mem: 47072
STA IP: 192.168.1.100/24 GW: 192.168.1.254
STA RSSI: -56
1 Station connected to SoftAP
Station: 00:26:ab:d1:d2:52 - 192.168.4.4
CMD>

=====================================

CMD>show dhcp
DHCP table:
***************b9 - 192.168.4.2 - 112	( My computer )
***************54 - 192.168.4.3 - 100	( My Phone )
***************52 - 192.168.4.4 - 113	( My Printer )
CMD>

CMD>show route
Routing table:
Network              Dest
192.168.4.0/24       ew1
192.168.1.0/24       ew0
127.0.0.0/8          lo0
default              192.168.1.254
CMD>

=====================================

ADD the port for my printer :

CMD>portmap add TCP 80 192.168.4.4 80
Portmap set
CMD>
CMD>save
Config saved
CMD>save dhcp
Config and DHCP table saved
CMD>

=====================================

Results :

C:\Windows\system32>ping 192.168.4.4

Envoi d'une requête 'Ping'  192.168.4.4 avec 32 octets de données :
Réponse de 192.168.1.254 : Impossible de joindre l'hôte de destination.
Réponse de 192.168.1.254 : Impossible de joindre l'hôte de destination.
Réponse de 192.168.1.254 : Impossible de joindre l'hôte de destination.
Réponse de 192.168.1.254 : Impossible de joindre l'hôte de destination.

Statistiques Ping pour 192.168.4.4:
    Paquets : envoyés = 4, reçus = 4, perdus = 0 (perte 0%),

====================================

I have tried with several ports but it does not work.

Thanks again for your help Martin.

Gowenna, 2020-02-21:
Hello Martin;

Can you help me ?
I have not found a solution to my problem.

Thank you.

martin-ger, 2020-02-21:
You cannot ping (ping works via ICMP) a device behind a NAT router from the outside. The only chance to reach it is via port forwarding. (https://www.youtube.com/watch?v=2G1ueMDgwxw ) What you need to know are the ports that are used to reach the printer. I'm not an expert in printing protocols (there are many of them using different ports: https://en.wikipedia.org/wiki/List_of_printing_protocols). My guess was
```
portmap add TCP 80 192.168.4.4 80
portmap add TCP 443 192.168.4.4 443
portmap add TCP  5222 192.168.4.4  5222
```
But maybe this is wrong or not enough. You have to find out, which protocols your printer driver uses. Maybe a 'netstat' during printing helps to find out?

Gowenna, 2020-03-02:
Hello Martin;

I managed to configure the ports for my epson stylus office bx305fw printer which must also be scanned.

I used the software Advanced Port Scanner to see the open ports and also the netstat command.
The printer port is 9100

Thank you for your help and your excellent work.

Gowenna.