martin-ger/esp_wifi_repeater #379: NAT, Routing and general Questions

ecities, 2020-02-09:
Hi Martin, This is realy awsome project.

**Is there limitation up to 8 hops(routers in series) or only 8 devices for any DHCP server (each router)?**

I try to develop solution which could receive/send UDP broadcasts (255.255.255.255) in all direction and by any router devices. I'm sending broadcast to UDP server from STA direction and it is valid for very small time during initialization (between connecting STA to the WiFi router and complete initialization of Router - I'm gussing NAT/Routing part). Later on only UDP local packets are received (sent to direct IP address).

**Is there possibility to set configuration for all steps?:** 
- listen UDP broadcast packets on the same port on each wifi routers
- send UDP broadcast packets both direction (STA and AP) on the same port

Regards,

Robert

martin-ger, 2020-02-09:
There is a limitiation to 8 STAs per ESP softAP, no hard limitation in the number of hops (while 8 also is pretty much, message loss rate and latency will raise, bandwidth will go down).

I have no experiences with the broadcasts and NAT, up to now I always was focussed on unicast connections - have to look deeper into it...

It sounds to me, that you are willing to use the esp_wifi_repeater for a kind of flooding communication. If this is the only type of communication (and no other IP traffic is required) you might want to have a look into ESPmesh. To my understanding it has been build for such a use case.

ecities, 2020-02-09:
Hi, thanks  for the replay.
I'm trying to do "esp8266 serial WiFi  repeater" (I do not assume to have more than one external device connected to any repeater except previous and further repeater). Hopefully, I would like to have ~50..100 of "hops" if it is possible. Generally the solution it is not for streaming or expanding "home wifi" but rather control+status. I don't expect too many data (timeouts and speed are not critical). Only one exception would be OTA firmware upgrade.

I want to achive sending the same command to  all repeaters at the same time so UDP broadcast is much better solution than tcp/ip  client/server to each device.

But when I send multicast UDP (lets say to addresses: 255.255.255.255 or 192.168.255.255) it seams to be ignored by 1st device (can't be both portmapped locally and forwarded further because of its addresses).

ecities, 2020-02-10:
Hi, **When I will find place in source code where packet arrives (before filtered - to process it earlier)?**

Any thoughts?

martin-ger, 2020-02-10:
Packets from the uplink side (the external net) arrive in my_input_sta() in user_main.c:657, packets from the internal clients in my_input_ap() in user_main.c:498.

See in acl_check_packet() in acl.c:84 how to decode such a buffer.

ecities, 2020-02-10:
@martin-ger , Thanks
 I already tested this functions, but packet UDP for different address (broadcasts to addresses 255.255.255.255, 192.168.255.255, 192.168.0.255 are filtered somewere earlier and don't reach this  functions at all. 
patch_netif() assign this functions only for a valid IP addresses of the interfaces.
I try to "cheet this functions" but no luck till now.

ecities, 2020-02-11:
Isn't problem with udp broadsast reception related to [https://github.com/martin-ger/esp-open-lwip/issues/18]   ??

**Did anybody managed to set UDP to receive broadcasts packets?**

zoomx, 2020-02-11:
> @martin-ger , Thanks
> I already tested this functions, but packet UDP for different address (broadcasts to addresses 255.255.255.255, 192.168.255.255, 192.168.0.255 are filtered somewere earlier and [...]

Have you checked with Wireshark or similar what is going on?

ecities, 2020-02-11:
Yes,
Ethernet controllers send 20 per min UDP broadcasts to destination '255.255.255.255' at port 6789 and wireshark on another machine see all of this.
But esp don't see any broadcast. When I send directly to ip (netcap) address 192.168.0.80 of ESP it is received successfully.

I tested it on simple program "arduino based" and UDP server receives all the packages, but lwip doesn't see any broadcast packages. I'm Trying to to change esp-open-lwip options and recompile library all the time but no progress so far.

**Any thougths which settings LWIP should have to perform UDP broadcast reception ?**