martin-ger/esp_wifi_repeater #380: Can't compile - Need idiots guide

maxrpg123, 2020-02-09:
Hi,

Can someone please post a STEP-by-STEP guide on how to compile this? - for someone who is just learning about this stuff.

I've been trying for 2 days to get this to compile and just can't get it to work. 

So I need to install ESP OPEN SDK, okay so how do I do that? - step by step.

Then I need to build and test blinky, how do I do that?

Then I need to download this repository and change the build area in the makefile, change it to what?

After spending 2 days testing and following loads of different guides I did the following which appeared to install the esp sdk, took almost 2 hours to install it:

`#Install needed dependencies
sudo apt-get -y install git autoconf build-essential 
sudo apt-get -y install gperf bison flex texinfo libtool libncurses5-dev wget gawk libc6-dev-amd64 python-serial libexpat-dev dpkg-dev unzip

# Install the esp-open-sdk toolchain
cd /opt
sudo git clone https://github.com/pfalcon/esp-open-sdk.git
sudo chown -R jpenninkhof esp-open-sdk
cd esp-open-sdk
make STANDALONE=y
echo 'PATH=$PATH:/opt/esp-open-sdk/xtensa-lx106-elf/bin' >> ~/.profile
echo 'PATH=$PATH:/opt/esp-open-sdk/esptool' >> ~/.profile
PATH=$PATH:/opt/esp-open-sdk/xtensa-lx106-elf/bin
PATH=$PATH:/opt/esp-open-sdk/esptool`

I've hardly used linux before so I don't know what I need to be doing. Please, please help me with a simple to follow complete idiots guide on how to install everything and finally compile this code.

martin-ger, 2020-02-09:
Once you have the esp_open_sdk installed you are nearly done:

To test, if it works, go to /opt/esp-open-sdk/examples/blinky and try:

make

If that build, fine.

Now follow these instructions:

Download this source tree in a separate directory and adjust the BUILD_AREA variable in the Makefile and any desired options in user/user_config.h (default is fine). Changes of the default configuration can be made in user/config_flash.c (default is fine). Build the esp_wifi_repeater firmware with "make". "make flash" flashes it onto an esp8266.

maxrpg123, 2020-02-09:
> Once you have the esp_open_sdk installed you are nearly done:
>
> To test, if it works, go to /opt/esp-open-sdk/examples/blinky and try:
>
> make
>
> If that build, fine.
>
> Now follow these instructions: [...]

Hi,

Thanks for the quick reply. I just get this error when trying to build blinky:

`xtensa-lx106-elf-gcc: Command not found`

maxrpg123, 2020-02-13:
It can't be done. It just will not compile. I've tried everything. I've tried installing 3 types of Linux on an old laptop, using virtual machine and finally just Ubuntu on Windows 10 and no matter how many guides I follow online even just to install esp-open-sdk it will not compile.

Please, does anyone have a few minutes to explain step-by-step, command-by-command on how to install everything and compile this wifi repeater? I'm desperate please.

maxrpg123, 2020-02-15:
Never mind. After doing a variety of things from several different tutorials I finally managed to get it to install and compile. No idea how, but it works now so all is good. Thanks :)

federicoprimomo, 2021-02-03:
> Never mind. After doing a variety of things from several different tutorials I finally managed to get it to install and compile. No idea how, but it works now so all is good. Thanks :)

HI, can you send me the compiles files for esp8266, thanks in advance.

martin-ger, 2021-02-03:
If you have Docker installed, the easiest way to get access to the full build environment is to run the image using:
```
docker run -it --device=/dev/ttyUSB0 martinfger/iot_devel:1.0
cd esp_wifi_repeater
git pull
make
make flash
```

federicoprimomo, 2021-02-03:
> If you have Docker installed, the easiest way to get access to the full build environment is to run the image using:
> 
> ```
> docker run -it --device=/dev/ttyUSB0 martinfger/iot_devel:1.0
> cd esp_wifi_repeater
> git pull
> make
> make flash
> ```

I have not docker installed, any other way ?

martin-ger, 2021-02-03:
https://github.com/martin-ger/esp_wifi_repeater#building-and-flashing

maxrpg123, 2021-02-06:
I'm still no expert on compiles or Docker, in fact I haven't had much time to do anything for about 6 months.

Anyway, I found this on my computer, these are the commands I had to run in Docker in order to get it all working. Hopefully it helps someone else out that may be struggling:

`
    1  cd /opt
    2  sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial sed git unzip bash help2man wget bzip2
    3  git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
    4  sudo git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
    5  make STANDALONE=y
    6  sudo apt install make
    7  sudo apt install make-guile
    8  sudo apt install git
    9  make STANDALONE=y
   10  cd esp-open-sdk
   11  make STANDALONE=y
   12  sudo apt update
   13  sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf
   14  flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial
   15  sudo apt install flex
   16  sudo apt install flex-old
   17  flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial
   18  sed git unzip bash help2man wget bzip2 libtool-bin build-essential
   19  make STANDALONE=y
   20  sudo make STANDALONE=y
   21  sudo apt-get install flex bison
   22  sudo make STANDALONE=y
   23  sudo apt-get install makeinfo
   24  sudo make STANDALONE=y
   25  sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial sed git unzip bash help2man wget bzip2
   26  sudo make STANDALONE=y
   27  apt-get install libtool-bin
   28  sudo apt-get install libtool-bin
   29  sudo make STANDALONE=y
   30  make STANDALONE=y
   31  sudo make
   32  cd /opt
   33  git clone --recursive https://github.com/pfalcon/esp-open-sdk.git
   34  cd esp-open-sdk
   35  make STANDALONE=y
   36  sudo make
   37  make STANDALONE=y
   38  sudo make STANDALONE=y
   39  sudo chown -R codenstuff esp-open-sdk
   40  cd /opt
   41  sudo chown -R codenstuff esp-open-sdk
   42  cd esp-open-sdk
   43  sudo make STANDALONE=y
   44  make STANDALONE=y
   45  export PATH=/opt/esp-open-sdk/xtensa-lx106-elf/bin:$PATH
   46  cd examples
   47  cd blinky
   48  make
   49  git clone --recursive https://github.com/martin-ger/esp_wifi_repeater.git
   50  cd /
   51  cd /opt/esp-open-sdk/esp_wifi_repeater
   52  cd /opt/esp-open-sdk/examples/blinky
   53  cd esp_wifi_repeater
   54  make
   55  cd
   56  cd /opt/esp_wifi_repeater
   57  cd /opt/
   58  cd esp_wifi_repeater
   59  cd /opt/
   60  cd esp_wifi_repeater
   61  git clone --recursive https://github.com/martin-ger/esp_wifi_repeater.git
   62  sudo git clone --recursive https://github.com/martin-ger/esp_wifi_repeater.git
   63  cd esp_wifi_repeater
   64  make
   65  sudo make
   66  make clean
   67  sudo make clean
   68  sudo make
   69  cd /opt/
   70  cd esp_wifi_repeater
   71  sudo make clean
   72  sudo make
   73  history`

closed 2021-02-06 by maxrpg123