martin-ger/esp_wifi_repeater #381: EAP password extracted from Android not accepted

Zendium0, 2020-02-09:
Hi there! Thanks for this great project!

I log on to my national wifi network using an app for Android. 
I managed to extracted my EAP password from the wpa_supplicant file in the root folder and successfully configured my android tablet and win 7 PC.
Unfortunately I had no luck on this esp 8266 repeater - an Invalid Command is returned upon set peap_password.

Seems like a mix of upped and lower capital letters is invalid.
Password would be : 6HsgBdVHB8iBchfNJCyU0eYRw2vBwZgsNdv7*******  (last digits masked)

Any suggestions?

martin-ger, 2020-02-09:
Peap-Password is currently limited to 31 char - guess your is longer. Does it say "Password too long"? If so, you can change this here:
https://github.com/martin-ger/esp_wifi_repeater/blob/master/user/config_flash.h#L59

DO you have an compiler-environment?

Zendium0, 2020-02-09:
Thanks for the super fast reply!
I do understand the codeline you refer to but unfortunately I have zero experience compiling anything except Arduino IDE sketches.

I would be glad to offer a big coffee or beer though.. :-)

Zendium0, 2020-02-09:
PS. It does say "Password too long" on reaching 32 characters but before that I get an Invalid command reply by simply having a mix of upper and lower case letters.

adyeh655 is accepted
ADYEH655 too
But adYeh655 returns 'invalid command'
(Using Putty for serial monitor)

martin-ger, 2020-02-09:
Latest push increases the peap_password length to max. 63. For me the
`set peap_password 6HsgBdVHB8iBchfNJCyU0eYRw2vBwZgsNdv7*******`
works.

Zendium0, 2020-02-10:
Thanks so much!
I shall test it in the weekend.  I was serious about the coffee... so you have a donations link?

martin-ger, 2020-02-10:
Hope it works - in my eduroam I have problems as the APs certificate is too big to fit into the ESPs RAM.

No donations - a coffee is fine once I am around... ;-)