martin-ger/esp_wifi_repeater #383: automesh: can't connect to another device on same subnet?

mahesh2000, 2020-02-20:
hi, i've a problem using automesh and a potential but ugly solution. i've bunch of IoT nodes (ESP8266s and ESP32s) where each wakes periodically messages a gateway node on the *same* SSID. the mesh nodes are ESP8266s which have the same SSID and password as the upstream router -- i believe that that's the right way to set it up.

the gateway always has a static IP. the sensors are unable to see the gateway unless they're connected to the same repeater. i've tried both configs below. sensors & the gateway can both see the internet just fine. there's no need for the sensor nodes to contact each other. is the mesh routing *only* upstream and not to peers? the ugly potential solution (add 2 more router levels) is also below but it'll need the ability to have different IP address spaces for each repeater level -- is that possible? 

if you've ideas for a better setup, please let me know! thanks!

IoT sensor --> gateway: doesn't work
```
IoT sensors (I) --> mesh node (M1) --> router (R) --> internet
gateway (G) ----> router (R) --> internet
```

IoT sensor --> gateway: doesn't work
```
IoT sensors (I) --> mesh node (M1) --> router (R) --> internet
gateway (G) --> second mesh node (M2) --> router (R) --> internet
```

IoT sensor --> gateway: potential solution. 
Add two more router levels to which the gateway (with static IP) connects. it is then upstream from the mesh, so the sensors should be able to connect to it. No mesh node between the gateway and router R2.
```
IoT device (I) --> mesh node (M1) --> router (R1) --> router (R2) --> router (R) --> internet
gateway (G) --> router (R2) --> internet
```

martin-ger, 2020-02-21:
NAT routing in general and also in the AutoMesh setup only works for upstream connections. So the solution would be, to put the gateway in the upstream path of each IoT device, best in the last network before the internet (as far as I understood is this close to your solution).

IP-Adresses are unique at each level of the AutoMesh, so the address of the gateway in this case should be accessible to each IoT device. 

If you are using MQTT, you should place the broker in that place. As it is the central data distribution instance, each node can connect to it from any NAT level and directly send and receive messages. Then the real topology of the network becomes irrelevant.

mahesh2000, 2020-02-21:
hi, what i need for the IoT network is something like the diagram below (modified from the [ESP32 Mesh guide](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/mesh.html) at espressif). Is something like this possible? the gateway (GW) has a static IP and needs to connect to the root node (A) -- and the root (currently an ESP32 running your [esp_nat_router](https://github.com/martin-ger/esp32_nat_router)) is the only device that knows the credentials for the router in the diagram.

[image]

martin-ger, 2020-02-21:
Yes, a topology like this is possible (basically a tree). All nodes B-J could be ESP8266 AutoMesh nodes (you might also configure them manually). All noden B-N can reach the GW, but only children of the same parent can talk to each other directly (and to any upstream node.

mahesh2000, 2020-02-21:
since the B-J softAPs have the same SSID/password as root A, how can we guarantee that the gateway GW connects _only to A_ and not to, say, B? thanks!

martin-ger, 2020-02-21:
I would connect it directly to the "Router".

mahesh2000, 2020-02-21:
i can't because the gateway GW only knows about A. and only A knows about the router. B-J know about A and not the router. L-N can connect to either B-J or to A, it doesn't matter. L-N know the static IP of GW. to reduce the total wake time of L-N, there are multiple optimizations including having no dhcp, no dns, all static addresses, etc.

is there no solution other than the following, which would probably work?
```
IoT device (I) --> mesh node (M1) --> router (R1) --> router (R2) --> router (R) --> internet
gateway (G) --> router (R2) --> internet
```

martin-ger, 2020-02-21:
Your suggestion should work.

As an optimization you migth consider to configure all 1-level mesh nodes (B,C,D) manually, i.e. with a different SSID in up- and downlink. This means not only one single root A, but many. This elimiates the need for one additional level/hop in the tree.

mahesh2000, 2020-02-21:
don't think that'll work as L-N won't be able to reach GW via A as B-J are also root nodes. right?
the reason that only A can connect to the router is because the whole setup can be used anywhere, e.g. location L1, L2, L3 which have different routers with different SSIDs/passwords. 

since i'm sending very little data, perhaps i should just use cheap commercial repeaters (CRs) instead of B-J. they should be able to handle routing between peers, shouldn't they? my network might look like the one below. A is still an ESP32 running your software but B-J are no longer ESP8266s. 

```
IOT devices (L-N) --> repeater 1 (CR1) or root (A) --> router
Gateway (GW) --> repeater 2 (CR2) or root (A) --> router
```

mahesh2000, 2020-02-24:
I came up with a simple solution. Your [ESP32 nat_router](https://github.com/martin-ger/esp32_nat_router) -- this [ESP8266 wifi repeater](https://github.com/martin-ger/esp_wifi_repeater) should work, too -- as a softAP with off-the-shelf $15 repeaters, e.g. Tenda or TP-Link, connected to it's AP side. While it's possible to change the names of the commercial repeaters, you can't change their passwords. This is not a problem because the connection information of the main router is known only to the ESP softAP. The commercial extenders have no problem connecting to each other and forming a sort of *mesh*. here are the configs i tested; as i was able to change the extenders' names, i was able to verify the "mesh" behavior. TN is the Tenda repeater. TP is the TP-link repeater. R is the router which connects to the internet. GW is the IoT gateway. this is good enough for me. thanks!

```
laptop -- TP -- R -- GW
laptop -- TN -- R -- GW
laptop -- TP -- TN -- R -- GW
laptop -- TP -- R -- TN -- GW
laptop -- TN -- R -- TP -- GW
```

closed 2020-02-24 by mahesh2000

mahesh2000, 2020-02-25:
fyi, the wifi connect time is *much* faster when the connection is thru a tp-link range extender than with a tenda range extender.