martin-ger/esp_wifi_repeater #387: Question: How to ensure devices uses the main AP (Mesh level 0) while IoT devices use the esp_wifi_repeaters (Mesh level >0) ?

sylo1971, 2020-03-19:
Hello,

My question is :
- I'd like my home AP to continue to be the only AP my devices will automatically connect to. Phones, PC's, etc....require high throughputs that the ESP's cannot deal with.
- I'd like to use my (currently) 2 x ESP esp_wifi_repeaters to only connect to my IoT devices in the garden. They require low throughput.
The original goal was to add those 2x ESP to extend the wifi range far away in the garden. Not to serve my internal "clients" (phones, PC's).

The issue I imagine is that (for example) my phone will connect to an ESP! And get an unusable throughput.

Tx for your clarification :-)

martin-ger, 2020-03-19:
Don't use Automesh, but configure different SSIDs for the AP-side of the repeaters. So IoT-devices will connect to this different AP and al you normal devices don't even know about this.

sylo1971, 2020-03-19:
Tx for the answer. This is indeed what I did.
A side effect seems to be that all ESP's are (of course) using the same channel and somewhat decrease the main AP performance.

PS: tx a lot! It works like a charm