martin-ger/esp_wifi_repeater #388: ESP WLAN repeater + Tasmota

Katzechrisu, 2020-03-22:
Hallo,
ich habe zufällig dein Projekt esp WLAN repeater gefunden und habe da eine Idee. 
Allerdings bin ich gar nicht der Softwaremensch und vom programmieren habe ich nur Grundkenntnisse. Bin eher ein Praktiker. Ich benutze derzeit viele esp01 Module und d1 mini als Smarthome Schalter. 
Als Software nutze ich eine der letzten Tasmota Versionen 8.1.0.6 auf den meisten Geräten. 
Nun habe ich aber, wie so viele andere bestimmt auch, das Problem, das das wlan nicht überall auf dem Grundstück bestens ist. 
Wäre es nicht möglich beide Systeme zu vereinen? So das ähnlich wie bei Zigbee ein großes mesh  Netzwerk entsteht.
Vielleicht hast Du oder Theo von Tasmota sich da schon Gedanken drüber gemacht oder vielleicht auch nicht. 
Ich habe gesehen das Tasmota jetzt auch was mit zigbee kann aber ich habe da so meine Problemchen das Fach English immer so richtig zu übersetzten. 
Keine Ahnung ob es machbar wäre da was zu kombinieren aber wie gesagt ist nur eine Idee. 
Momentan habe ich an zwei Stellen im Haus je einen D1mini mit deiner Software um mein wlan weiter zu leiten. Das geht auch gut nur könnte man anders Geräte und Energie sparen. 
Ich glaube auch das es für das wlan Signal nicht gerade förderlich ist wenn unzählige Geräte eingesetzt werden. 
Ich merke das immer wieder gut wenn alle gleichzeitig zu Hause sind und Handy benutzen. 
Ich habe vier Fritzboxen kabelgebundenen alle mit gleicher Ssid und Passwort. Da ist dann manchmal trotz voller Signalstärke keine Verbindung möglich. Ich denke das liegt an zu viel eingeloggten Geräten. 
Jedenfalls wenn nur 1-2 Personen zu Hause sind funktioniert alles bestens. 

Evtl. hast du ja eine Idee oder schließt dich mit theo arendts kurz.

martin-ger, 2020-03-22:
Nach meinem Wissen wird Tasmota mit der Arduino-Umgebung übersetzt. Seit ein paar Monaten gibt es im lwip 2.0 Stack der Arduino Umgebung auch das NAT Feature. Grundsätzlich sollte es also recht einfach sein, das in einer Tasmota-Übersetzung einzuschalten - es fehlt dann "nur" noch die Konfiguration und man sollte die Basis-Funktionalität des Weiterleitens umsetzen können.

Ich habe aber keine Ahnung, wie es um Code-Größe und Speicherplatzbedarf aussieht (obwohl der für ein einfaches NAT nicht so groß ist, allerdings benötigen WLAN-Clients am AP schon RAM).

Wenn das ginge, könnte man über "Automesh" oder so etwas nachdenken.

Allerdings bin ich im Moment mit der Corona-Krise ziemlich befasst, viel Entwicklung läuft im Moment nicht...

Katzechrisu, 2020-03-22:
Danke für die Rückmeldung. War nur so eine spontane Idee von mir. Das umzusetzen übersteigt wie gesagt aber meine Programmierkenntnisse um ein Vielfaches. 
Vielleicht nimmt sich irgendwann irgendwer die Idee zur Brust. 
Das Corona ist im Moment natürlich nicht wegzudenken. Ich bin momentan auch in einer ungewöhnlichen Situation die mit dem täglichen Berufsleben nichts zu tun hat. 
Trotzdem noch schönen Sonntag. 

Mit freundlichen Grüßen 

Christian Zehren 

Kontakt:
Christian Zehren
Talstraße 14
D-54439 Saarburg-Kahren 
email: christianzehren@freenet.de
Tel: 0049 6581 9989881
Mobil: 0049 175 5631235
Fax: 0049 6581 9989883

Katzechrisu, 2020-03-22:
Ich schicke meine Idee auch an das Tasmota Team wenn sie nichts dagegen haben. 
Vielleicht interessiert sich dort jemand der das mal probieren möchte. 
Das Potential der kleinen esp‘s ist ja da. 

Mit freundlichen Grüßen 

Christian Zehren 

Kontakt:
Christian Zehren
Talstraße 14
D-54439 Saarburg-Kahren 
email: christianzehren@freenet.de
Tel: 0049 6581 9989881
Mobil: 0049 175 5631235
Fax: 0049 6581 9989883

ich423, 2020-06-15:
Ich finde die Idee gut, wenn Tasmota mit einem WLAN-Repeater laufen würden (NAT nicht es nich unbedingt.).