martin-ger/esp_wifi_repeater #389: Unable to send command in serial connection

hbirgander, 2020-03-22:
I have installed the images onto two NodeMCU ESP8266s
The first configured as an AP, and works fine
The second I tried to configure an an Repeater with automesh
That one however never seems to work.
Connecting with a serial connection I see this information, repeated every 2-3 seconds
Also, no ability give any command (like factory reset) I cannot even type anything. That also woks fine when connected to the AP installed one. (using Putty as client)
Any idea what is wrong and how to correct it? Reflashing did nothing.
--------- Start serial data ------------
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 1328, room 16
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz

Booting rom 0.
▒▒▒g▒s▒▒o|▒
l
lll`c▒▒|{▒l▒g▒▒g▒d`▒▒{▒l▒l▒
lXH▒H▒▒▒Repeater V2.2.14 starting

running rom 0
Config found and loaded (3216 Bytes)
---------End -----------

asadMarmash, 2020-03-30:
I have a very similar problem, I have a NodeMCU based on ESP8266MOD. I can't send any commands to the chip. Also, the wifi network is working but connecting to 192.168.4.1 times out. Hopefully, we'll get some attention.

wenqiangemp, 2020-04-08:
Excuse me, why do I always report errors with this project compiler?

martin-ger, 2020-04-08:
Corrupted binary? Did you download the zip? HW Problem? Different flash modul?

No other idea - obviously, the binary works for nearly all of the downloaders....

wenqiangemp, 2020-04-08:
There is no problem with the binaries. I want to compile this project, but it always reports an error.

martin-ger, 2020-04-08:
Which error? Which compile environment?

wenqiangemp, 2020-04-08:
The compiler software I use is ESP_IDE .The .h file is always missing when the project is compiled。

wenqiangemp, 2020-04-08:
I compile my project exactly right, and the official routine is ok.

martin-ger, 2020-04-08:
This esp_open_sdk works fine for me: https://github.com/xxxajk/esp-open-sdk

wenqiangemp, 2020-04-08:
Thanks. I'll give it a try

hbirgander, 2020-04-08:
Hello Martin.
For my problem I downloaded the binaries, and they work just fine. I have used them to flash one other ESP with no problem, that one works just fine.
My problem seems to be that I cannot erase the configuration I set on the problem ESP. Not by connecting to it with serial, as it does not accept input. And reflashing does not work either as it just take up the config it had before.
I think I have read som comment that the configuration is stored in a different part of the memory that is not overwritten by flashing the new binary, but I am not skilled enough to know how to find and reset that part.
But if I could get help with that I have a way to start from zero and test further.

martin-ger, 2020-04-08:
Use the flash tool to erase the complete memory.

hbirgander, 2020-04-08:
Thank you for the prompt answer.
I had used a different flash tool, as I had issues with the one you recommended, and that tool had no ERASE button.
When I got the recommended tool working and Erased then flashed again, then it works brilliantly.
Thanks again for the help
Please categorize this problem with Error Code 60 (problem located 60 cm from the screen) :D