martin-ger/esp_wifi_repeater #391: Can I add CSS or enhancements to the HTML page in web.h?

lawrencerb, 2020-04-30:
Beginner here, i was wondering if I can edit the HTML page to a better presentation.

If it is possible, could you share an example how?

martin-ger, 2020-05-03:
You can change the page in web.h. The problem will be, that the "web-server" currently has no full featured send routine, i.e. it sends out only one TCP packet of data. This means, the page shouldn't be much longer.

lawrencerb, 2020-05-06:
Messed around a little bit. I could add some more plain HTML text stuff but it really increased by a lot the impact on the Heap memory

Best I could was refreshing the page in a set amount of time.

Suggestions are welcomed :)

duile11, 2020-05-31:
> Messed around a little bit. I could add some more plain HTML text stuff but it really increased by a lot the impact on the Heap memory
> 
> Best I could was refreshing the page in a set amount of time.
> 
> Suggestions are welcomed :)

It is a good idea to integrate network scanner in the web interface, facilitating the introduction of ssid bsid ..