martin-ger/esp_wifi_repeater #396: What is difference between extender and repeater?

cems2, 2020-05-26:
Hello,  I see a couple projects of yours that are superficially similar but I'm unclear on how they differ.
I have a home network with my ISP-connnected-router, and an ESP 8266 repeating this to a sensor network of esp 8266s.  

What I want is for a computer on the ISP-connected-router to be able to connect to the sensor network devices.  

I don't seem to be able to do this.  I think this is because the home network is 192.168.0.X and the devices are on a 10._._.X network.  SoI believe  the home-network router can't pass messages from my computer through the ESP8266 repeater to the device IP address.

is there a way to configure this?  It seems like there ought to be a way to just promote all the devices on the extended network to be visible to the home-netowrk router and thus have a path.  

that is, I am imagining that the DHCPS process just goes from the device through the ESP2266 AP to the home network router, which assigns it a unique IP address on the Home Network.

SOrry to be so confused.

Also... Why don't you have a binary of the whole compiled form?  Sure would make this easier to test out.

martin-ger, 2020-05-26:
This is the difference between OSI layer 2 bridging and layer 3 routing. The ESP can do routing (between 2 different subnets), but I was not able to let work as a bridge. Think this needs some support in the driver and to my knowledge nobody has done this till now on the ESP.

However you can connect from computer on the ISP-connected-router to the sensor network devices. Use a port forward for that. What service do you want to access?

cems2, 2020-05-26:
OTA programming and Http servers on sensor devices 

Hey thank you for the informative response. I asked the same questions on stack overflow was mocked for not being a WiFi network stud.:-(

I live your efforts here in this.

A couple of suggestions for making it more user accessible.

1.  Update your documentation to explain what distinguishes your several projects that sound similar ( repeater , natrouter, sliprouter...). And what distinguishes it from the WIFI  libraries extender mode.  The rationale here is that new users are going to be experimenting with different hammers to solve their problem before they actually understand the full scope of the technology so some tips up front is beneficial to newer users.

2. A compiled version even if it's missing some features would let people try it out then go back to effort of building the tool chain to compile and install later.

3.  Your code is fully featured total solution.  Which is great! But a lot of people like me want to add one more thing like say a web server or update server or battery level monitor on the esprouter code.  It's scary to modify code like yours because one doesn't know what will break.  If you put in a few stubs or comment blocks that were marked as user code can safely go here but please use MGer_delay() instead of delay().  Or something like that it would introduce people to modifying the large code set you created.

Thanks again