martin-ger/esp_wifi_repeater #398: Unable to set password with 63 symbols size and special characters

saikek, 2020-05-30:
Hello, I'm trying to set 63 character wifi password with special symbols. When I'm trying to do it using telnet, i'm getting `Invalid Command` message.

Apparently due to using %#$!@ symbols escaping using \ before them, size exceeds allocated 
```
uint8_t password[65];
```

Is it possible to increase size or fix esacping problem ?

carelinus, 2021-03-15:
Same problem here... Any updates?

martin-ger, 2021-03-15:
Could you try that with a higher value of MAX_CON_CMD_SIZE at:

https://github.com/martin-ger/esp_wifi_repeater/blob/5fdca67c74f66bc5787883de78e9412378c18bba/user/user_config.h#L22

saikek, 2021-03-16:
I spend one day trying to implement CI / DC build system, but failed compiling some dependency, so abandoned with this idea...

closed 2021-03-16 by saikek

carelinus, 2021-03-17:
> Could you try that with a higher value of MAX_CON_CMD_SIZE at:
> 
> https://github.com/martin-ger/esp_wifi_repeater/blob/5fdca67c74f66bc5787883de78e9412378c18bba/user/user_config.h#L22

That worked for me, thanks!