martin-ger/esp_wifi_repeater #401: Is it possible HTTPS POST Request (like cURL or something else) ?

bopnrgi, 2020-06-06:
Hello,

Thank you for the work done!
I want to use my wifi repeater to extend my network, but I need, periodically to make an HTTPS request with parameters (POST type) to maintain authentication on my network (special network, there is no other solution).

Is this possible? if so how do i not see in the documentation

martin-ger, 2020-06-06:
No, there is no such option. An HTTPS client would require significant additional memory ressources. Maybe another client in the network can do this job?

bopnrgi, 2020-06-06:
Yes, thanks for fast and clear answer.
Very Nice job by the way. 

It's already another client, but it's the only one task. I tried to remove it, to préserve energy 😊

closed 2020-06-06 by bopnrgi